dentra/esphome-tion #1: add support tion 3s

glv-altx, 2022-02-07:
please :+1:

dentra, 2022-02-08:
I have some research in S3 protocol, but need more time to learn some more

dentra, 2022-02-09:
Could you please get the advertisement packet from your device? It can be easily done via nRF Connect mobile app. Just open it, tap scan, wait a moment, find device in the list and the last string is it. Something like `<0xFFFF> 0xXX XX XX XX XX XX XX XX 00 00 00`

xAlexo, 2022-04-04:
Нужно еще что то? У меня есть tion 3s

dentra, 2022-04-04:
@xAlexo сейчас в мастере есть экспериментальная поддержка, но большая проблема с сопряжением. без него невозможно получить состояние бризера, функции управления работают. Проверить/попробовать можно на версии ESPHome 2022.2.x (выше не будет пока собираться)

конфиг пока примерно такой:
```yaml
substitutions:
  name: "Tion 3s"
  mac_tion: 00:00:00:00:00:00

external_components:
  - source: github://dentra/esphome-tion@master

logger:
  level: VERBOSE
  logs:
    ota: DEBUG
    mdns: INFO
    wifi: INFO
    sntp: INFO
    esp-idf: INFO
    api.connection: INFO
    api: INFO
    button: INFO
    switch: INFO
    sensor: INFO
    select: INFO
    number: INFO
    climate: INFO
    text_sensor: INFO
    binary_sensor: INFO
    esp32_ble_tracker: INFO

esphome:
  on_boot:
    # ESP_BLE_SEC_ENCRYPT, ESP_BLE_SEC_ENCRYPT_NO_MITM, ESP_BLE_SEC_ENCRYPT_NO_MITM
    # для запрета BT сопряжения: static_cast<esp_ble_sec_act_t>(0)
    - lambda: id(tion_climate).set_ble_encryption(ESP_BLE_SEC_ENCRYPT);
    # разрешить изменять состояние без получение статуса
    - lambda: id(tion_climate).set_direct_write(true);

ota:
  on_begin:
    then:
      - lambda: id(ble_client_tion).set_enabled(false);

ble_client:
  - mac_address: $mac_tion
    id: ble_client_tion

button:
  - platform: template
    name: "$name Pair"
    on_press:
      lambda: id(tion_climate).pair();
    entity_category: config
  - platform: template
    name: "$name Reset Pair"
    on_press:
      lambda: id(tion_climate).reset_pair();
    entity_category: config

climate:
  - platform: tion_3s
    id: tion_climate
    ble_client_id: ble_client_tion
    name: "$name"
    buzzer:
      name: "$name Buzzer"
    temp_in:
      name: "$name Temp in"
    temp_out:
      name: "$name Temp out"
    filter_days_left:
      name: "$name Filter Days Left"
    air_intake:
      name: "$name Air Intake"
    update_interval: 60s
```

кнопки Pain и Reset Pair нужны, чтобы "играться" с программным  сопряжением.

xAlexo, 2022-04-04:
У меня сейчас на 2022.3.2 собрана прошивка для esp32, вроде без проблем
Получение данных и переключатели не работают

В логах вот так вылядит:
```
[15:05:54][D][switch:013]: 'Tion 3s Room 2 Buzzer' Turning ON.
[15:05:54][D][switch:037]: 'Tion 3s Room 2 Buzzer': Sending state ON
[15:05:54][D][tion-api-3s:116]: Write state
[15:05:55][D][switch:017]: 'Tion 3s Room 2 Buzzer' Turning OFF.
[15:05:55][D][switch:037]: 'Tion 3s Room 2 Buzzer': Sending state OFF
[15:05:55][D][tion-api-3s:116]: Write state
```

dentra, 2022-04-04:
@xAlexo тут есть момент, бризер ввести в режим сопряжения, дальше стартануть ESP. В случае 4s и lite этого достаточно, в случае 3s нужно еще бризеру послать спц команду, делается кнопкой Pair.

Логи мне нужны именно c уровнем VERBOSE, пример, чтобы было читаемо, я привел.

xAlexo, 2022-04-04:
1. Перевел бризер в режим соряжения (кнопка мигает)
2. Запустил esp
3. нажал pair
4. Подождал

Данных нет, вкл/выкл без реакции
```
[15:59:28][D][ble_client:047]: Found device at MAC address [**:**:**:**:**:**]
[15:59:28][I][ble_client:085]: Attempting BLE connection to **:**:**:**:**:**
[15:59:28][D][tion:093]: Bonding complete: YES
[15:59:28][V][ble_client:115]: [**:**:**:**:**:**] ESP_GATTC_OPEN_EVT
[15:59:29][V][ble_client:134]: cfg_mtu status 0, mtu 23
[15:59:29][V][ble_client:159]: [**:**:**:**:**:**] ESP_GATTC_SEARCH_CMPL_EVT
[15:59:29][I][ble_client:161]: Service UUID: 0x1800
[15:59:29][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x7
[15:59:29][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0xa
[15:59:29][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[15:59:29][I][ble_client:347]:  characteristic 0x2A04, handle 0x7, properties 0x2
[15:59:29][I][ble_client:162]:   start_handle: 0x9  end_handle: 0xffff
[15:59:29][I][ble_client:347]:  characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xb, properties 0x10
[15:59:29][V][ble_client:382]:    descriptor 0x2902, handle 0xc
[15:59:29][I][ble_client:347]:  characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xe, properties 0xc
[15:59:29][V][ble_client:142]: [**:**:**:**:**:**] ESP_GATTC_DISCONNECT_EVT
[15:59:31][D][tion-api-3s:116]: Write state
[15:59:31][V][tion:114]: write_data to 0xe: 3D.02.00.00.00.08.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
[15:59:32][D][tion-api-3s:116]: Write state
[15:59:32][V][tion:114]: write_data to 0xe: 3D.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
[15:59:52][D][xiaomi_ble:314]: Got Xiaomi LYWSD02 (**:**:**:**:**:**:
[15:59:52][D][xiaomi_ble:320]:   Humidity: 41.0%
[15:59:59][D][xiaomi_ble:314]: Got Xiaomi LYWSD02 (**:**:**:**:**:**):
[15:59:59][D][xiaomi_ble:317]:   Temperature: 25.2°C
```

xAlexo, 2022-04-05:
Перенёс esp в комнату с 3s
Подключить к адаптеру который точно может по usb выдать 3.4А
Включил сопряжение на 3s
Нажал pair

```
10:59:36 | [D] | [ble_client:047] | Found device at MAC address [**:**:**:**:**:**]
10:59:36 | [I] | [ble_client:085] | Attempting BLE connection to **:**:**:**:**:**
10:59:36 | [D] | [tion:093] | Bonding complete: YES
10:59:36 | [V] | [ble_client:115] | [**:**:**:**:**:**] ESP_GATTC_OPEN_EVT
10:59:37 | [V] | [ble_client:134] | cfg_mtu status 0, mtu 23
10:59:37 | [V] | [ble_client:159] | [**:**:**:**:**:**] ESP_GATTC_SEARCH_CMPL_EVT
10:59:37 | [I] | [ble_client:161] | Service UUID: 0x1800
10:59:37 | [I] | [ble_client:162] | start_handle: 0x1  end_handle: 0x7
10:59:37 | [I] | [ble_client:347] | characteristic 0x2A00, handle 0x3, properties 0xa
10:59:37 | [I] | [ble_client:347] | characteristic 0x2A01, handle 0x5, properties 0x2
10:59:37 | [I] | [ble_client:347] | characteristic 0x2A04, handle 0x7, properties 0x2
10:59:37 | [I] | [ble_client:161] | Service UUID: 0x1801
10:59:37 | [D] | [tion:073] | Discovering complete
10:59:37 | [I] | [ble_client:347] | characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xe, properties 0xc
10:59:37 | [V] | [tion:074] | TX handle 0xe
10:59:37 | [V] | [tion:075] | RX handle 0xb
10:59:37 | [V] | [tion:114] | write_data to 0xe: 3D.05.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
10:59:37 | [D] | [tion-api-3s:103] | Enter pair mode
10:59:37 | [D] | [tion_3s:063] | Pairing complete: YES
10:59:37 | [V] | [component:199] | Component esp32_ble_tracker took a long time for an operation (0.16 s).
10:59:37 | [V] | [tion:102] | write_char at 0xe complete 0x00
10:59:40 | [I] | [ble_client:075] | [**:**:**:**:**:**] Disabling BLE client.
10:59:40 | [V] | [ble_client:142] | [**:**:**:**:**:**] ESP_GATTC_DISCONNECT_EVT
10:59:42 | [D] | [ble_client:047] | Found device at MAC address [**:**:**:**:**:**]
10:59:42 | [I] | [ble_client:085] | Attempting BLE connection to **:**:**:**:**:**
10:59:42 | [D] | [tion:093] | Bonding complete: YES
10:59:42 | [V] | [ble_client:115] | [**:**:**:**:**:**] ESP_GATTC_OPEN_EVT
10:59:43 | [V] | [ble_client:134] | cfg_mtu status 0, mtu 23
10:59:43 | [V] | [ble_client:159] | [**:**:**:**:**:**] ESP_GATTC_SEARCH_CMPL_EVT
10:59:43 | [I] | [ble_client:161] | Service UUID: 0x1800
10:59:43 | [I] | [ble_client:162] | start_handle: 0x1  end_handle: 0x7
10:59:43 | [I] | [ble_client:347] | characteristic 0x2A00, handle 0x3, properties 0xa
10:59:43 | [I] | [ble_client:162] | start_handle: 0x9  end_handle: 0xffff
10:59:43 | [I] | [ble_client:162] | start_handle: 0x9  end_handle: 0xffff
10:59:43 | [I] | [ble_client:347] | characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xb, properties 0x10
10:59:43 | [V] | [ble_client:382] | descriptor 0x2902, handle 0xc
10:59:43 | [I] | [ble_client:347] | characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xe, properties 0xc
10:59:43 | [D] | [tion:073] | Discovering complete
10:59:43 | [V] | [tion:074] | TX handle 0xe
10:59:43 | [V] | [tion:043] | write_char_descr at 0xc complete 0x00
10:59:43 | [D] | [tion-api-3s:111] | Request state
10:59:43 | [V] | [tion:114] | write_data to 0xe: 3D.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
10:59:43 | [V] | [tion_3s:054] | Request state result: YES
10:59:52 | [D] | [esp32.preferences:114] | Saving preferences to flash...
```

Всё по прежнему

dentra, 2022-04-05:
теперь видно, что подключается к устройству, конектится, получает характеристики, отправляет спец команду сопряжения (`3D.05.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A`). в теории все "приплясывания" выполнены. дальше в теории бризер должен начать отправлять свое состояние на запрос `3D.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A`. Очень странно, что лог содержит не все сообщения, в частности отсвутсвует RX handle после TX handle, Register for notify и Registring for notify complete перед write_char_descr at 0xc complete 0x00

теоретически сейчас если в on_boot добавить id(tion_climate).set_direct_write(true), то команды управления будут посылаться невзирая на отсутсвие обратной связи

xAlexo, 2022-04-05:
id(tion_climate).set_direct_write(true) это то что в конфиге выше? тогда это уже есть
сейчас есть реакция на выключение, но нет реакции на включение и нет данных

Включение и выключение
```
[13:45:47][D][tion-api-3s:116]: Write state
[13:45:47][V][tion:114]: write_data to 0xe: 3D.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
[13:45:47][V][tion:102]: write_char at 0xe complete 0x00
[13:45:49][D][tion-api-3s:116]: Write state
[13:45:49][V][tion:114]: write_data to 0xe: 3D.02.00.00.00.08.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
[13:45:49][V][tion:102]: write_char at 0xe complete 0x00
```

dentra, 2022-04-05:
данных нет, т.к.возможно по каким-то причинам не проходит программный pair. но, с другой стороны, девайс остается видимым, а по информации из интернетов, это значит, что pair все-таки прошел.

dentra, 2022-04-05:
Если у кого-нибудь есть возможность скапчурить трафик мобильного приложения при сопряжении и первичном обмене данных (идеально в формате wireshark), был бы очень признателен.

dentra, 2022-05-04:
@xAlexo хорошая новость, запустили таки с @glv-altx работу 3S, обязательно требуется ESPHome 2022.4.0 или выше

xAlexo, 2022-05-07:
Из мастера пишет:
```
[temp_in] is an invalid option for [climate.tion_3s]. Please check the indentation.
[temp_out] is an invalid option for [climate.tion_3s]. Please check the indentation.
```

Закоментировал их, залил новую прошивку:
```INFO Successfully connected to esph01.local
[13:39:52][D][ble_client:047]: Found device at MAC address [**:**:**:**:**:**]
[13:39:52][I][ble_client:085]: Attempting BLE connection to **:**:**:**:**:**
[13:39:52][V][ble_client:124]: [**:**:**:**:**:**] ESP_GATTC_CONNECT_EVT
[13:39:52][V][tion:100]: Bonding complete: YES
[13:39:52][V][ble_client:115]: [**:**:**:**:**:**] ESP_GATTC_OPEN_EVT
[13:39:52][I][ble_client:220]: auth complete. remote BD_ADDR: c06a924f734b
[13:39:52][V][ble_client:225]: auth success. address type = 1 auth mode = 1
[13:39:52][V][ble_client:139]: cfg_mtu status 0, mtu 23
[13:39:53][V][ble_client:164]: [**:**:**:**:**:**] ESP_GATTC_SEARCH_CMPL_EVT
[13:39:53][I][ble_client:166]: Service UUID: 0x1800
[13:39:53][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x7
[13:39:53][I][ble_client:378]:  characteristic 0x2A00, handle 0x3, properties 0xa
[13:39:53][I][ble_client:378]:  characteristic 0x2A01, handle 0x5, properties 0x2
[13:39:53][I][ble_client:378]:  characteristic 0x2A04, handle 0x7, properties 0x2
[13:39:53][I][ble_client:166]: Service UUID: 0x1801
[13:39:53][I][ble_client:167]:   start_handle: 0x8  end_handle: 0x8
[13:39:53][I][ble_client:166]: Service UUID: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
[13:39:53][I][ble_client:167]:   start_handle: 0x9  end_handle: 0xffff
[13:39:53][I][ble_client:378]:  characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xb, properties 0x10
[13:39:53][V][tion:081]:   TX handle 0xe
[13:39:54][V][json:040]: Size after shrink 56 bytes
[13:39:54][V][json:040]: Size after shrink 176 bytes
[13:39:54][V][json:040]: Size after shrink 148 bytes
[13:39:54][V][json:040]: Size after shrink 80 bytes
[13:39:54][V][json:040]: Size after shrink 92 bytes
[13:39:54][V][json:040]: Size after shrink 404 bytes
[13:39:54][V][json:040]: Size after shrink 212 bytes
[13:44:45][I][ota:105]: Boot seems successful, resetting boot loop counter.
[13:44:45][D][esp32.preferences:114]: Saving preferences to flash...
```
Данных нет, на переключение не реагирует

xAlexo, 2022-05-07:
и set_direct_write тоже убрал:
`livingroom.yaml:5:21: error: 'class esphome::tion::Tion3s' has no member named 'set_direct_write'`

spijet, 2022-05-08:
Приветствую!

Попробовал подружить свой Tion 3S с платой ESP32-WROOM-32D DevKit V4. В зависимости от выбранного режима шифрования модуль либо ругается на `invalid auth, reason 0x66` и перестаёт ругаться после повторного сопряжения, либо молча соединяется, вроде бы как что-то запрашивает у бризера, но не получает ответ (либо не может его распарсить). 

<details><summary>Лог ESPhome при попытке сопряжения и изменения настроек</summary>
<p>

```
[18:07:05][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[18:07:35][D][button:013]: 'Tion 3S Pair' Pressed.
[18:07:37][D][button:013]: 'Tion 3S Reset Pair' Pressed.
[18:07:37][I][ble_client:075]: [f2:ed:73:31:4f:7d] Disabling BLE client.
[18:07:37][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 22
[18:07:38][D][button:013]: 'Tion 3S Pair' Pressed.
[18:07:39][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[18:07:39][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[18:07:39][V][ble_client:124]: [f2:ed:73:31:4f:7d] ESP_GATTC_CONNECT_EVT
[18:07:39][V][tion:100]: Bonding complete: YES
[18:07:39][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[18:07:39][I][ble_client:220]: auth complete. remote BD_ADDR: f2ed73314f7d
[18:07:39][V][ble_client:225]: auth success. address type = 1 auth mode = 1
[18:07:39][V][ble_client:139]: cfg_mtu status 0, mtu 23
[18:07:39][V][ble_client:164]: [f2:ed:73:31:4f:7d] ESP_GATTC_SEARCH_CMPL_EVT
[18:07:39][I][ble_client:166]: Service UUID: 0x1800
[18:07:39][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x7
[18:07:39][I][ble_client:378]:  characteristic 0x2A00, handle 0x3, properties 0xa
[18:07:39][I][ble_client:378]:  characteristic 0x2A01, handle 0x5, properties 0x2
[18:07:39][I][ble_client:378]:  characteristic 0x2A04, handle 0x7, properties 0x2
[18:07:39][I][ble_client:166]: Service UUID: 0x1801
[18:07:39][I][ble_client:167]:   start_handle: 0x8  end_handle: 0x8
[18:07:39][I][ble_client:166]: Service UUID: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
[18:07:39][I][ble_client:167]:   start_handle: 0x9  end_handle: 0xffff
[18:07:39][I][ble_client:378]:  characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xb, properties 0x10
[18:07:39][V][ble_client:413]:    descriptor 0x2902, handle 0xc
[18:07:40][I][ble_client:378]:  characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xe, properties 0xc
[18:07:40][D][tion:080]: Discovering complete
[18:07:40][V][tion:081]:   TX handle 0xe
[18:07:40][V][tion:082]:   RX handle 0xb
[18:07:40][D][tion-api-3s:105]: Enter pair mode
[18:07:40][V][tion:121]: write_data to 0xe: 3D.05.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5A (20)
[18:07:40][D][tion_3s:068]: Pairing complete: YES
[18:07:40][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.11 s).
[18:07:40][V][component:200]: Components should block for at most 20-30ms.
[18:07:40][V][tion:109]: write_char at 0xe complete 0x00
[18:07:42][D][esp32.preferences:114]: Saving preferences to flash...
[18:07:43][I][ble_client:075]: [f2:ed:73:31:4f:7d] Disabling BLE client.
[18:07:43][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 22
[18:07:46][D][button:013]: 'Tion 3S Reset Pair' Pressed.
[18:08:01][D][button:013]: 'Tion 3S Pair' Pressed.
[18:08:01][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[18:08:01][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[18:08:01][V][ble_client:124]: [f2:ed:73:31:4f:7d] ESP_GATTC_CONNECT_EVT
[18:08:01][V][tion:100]: Bonding complete: YES
[18:08:01][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[18:08:02][I][ble_client:220]: auth complete. remote BD_ADDR: f2ed73314f7d
[18:08:02][V][ble_client:225]: auth success. address type = 1 auth mode = 1
[18:08:02][V][ble_client:139]: cfg_mtu status 0, mtu 23
[18:08:02][V][ble_client:164]: [f2:ed:73:31:4f:7d] ESP_GATTC_SEARCH_CMPL_EVT
[18:08:02][I][ble_client:166]: Service UUID: 0x1800
[18:08:02][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x7
[18:08:02][I][ble_client:378]:  characteristic 0x2A00, handle 0x3, properties 0xa
[18:08:02][I][ble_client:378]:  characteristic 0x2A01, handle 0x5, properties 0x2
[18:08:02][I][ble_client:378]:  characteristic 0x2A04, handle 0x7, properties 0x2
[18:08:02][I][ble_client:166]: Service UUID: 0x1801
[18:08:02][I][ble_client:167]:   start_handle: 0x8  end_handle: 0x8
[18:08:02][I][ble_client:166]: Service UUID: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
[18:08:02][I][ble_client:167]:   start_handle: 0x9  end_handle: 0xffff
[18:08:02][I][ble_client:378]:  characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xb, properties 0x10
[18:08:02][V][ble_client:413]:    descriptor 0x2902, handle 0xc
[18:08:02][I][ble_client:378]:  characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, handle 0xe, properties 0xc
[... 172 lines cut ...]
    name: "$name Reset Pair"
    on_press:
      lambda: id(tion_climate).reset_pair();
    entity_category: config

climate:
  - platform: tion_3s
    id: tion_climate
    ble_client_id: ble_client_tion
    name: "$name"
    buzzer:
      name: "$name Buzzer"
    outdoor_temperature:
      name: "$name Temp (Outside)"
    filter_days_left:
      name: "$name Filter Days Left"
    air_intake:
      name: "$name Air Intake"
    update_interval: 60s
```

</p>
</details>

Если могу чем-то помочь — дайте знать. Посниффать BT-пакеты вряд ли смогу, т.к. использую iOS. :(

spijet, 2022-05-08:
После некоторых безуспешных игрищ с конфигом ESPhome я попробовал убрать хендлер `esphome.on_boot` вообще и выкрутить логгирование всех плагинов, связанных с BLE и `tion{,_3s,-api-*}`, в режим `VERY_VERBOSE`. Собрал прошивку, залил в плату и заметил, что в выхлопе-логе появились записи со структурой, отражающей *правильное* текущее состояние настроек бризера, и в HA начали появляться все текущие показатели. Управление тоже работает, иногда с задержкой в 1-2 секунды.

<details><summary>Конфиг, на котором всё заработало</summary>
<p>

```yaml
substitutions:
  name: "Tion 3S"
  mac_tion: "F2:ED:73:31:4F:7D"

external_components:
  - source: github://dentra/esphome-tion@master

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 115200
  level: VERY_VERBOSE
  logs:
    ota: DEBUG
    mdns: INFO
    wifi: INFO
    sntp: INFO
    esp-idf: INFO
    api.connection: INFO
    api: INFO
    button: DEBUG
    switch: INFO
    sensor: INFO
    select: INFO
    number: INFO
    climate: INFO
    text_sensor: INFO
    binary_sensor: INFO
    ble_client: VERY_VERBOSE
    tion: VERY_VERBOSE
    tion_3s: VERY_VERBOSE
    tion-api-3s: VERY_VERBOSE
    esp32_ble_tracker: INFO

# Enable Home Assistant API
api:
  password: "<SNIP>"

ota:
  password: "<SNIP>"
  on_begin:
    then:
      - lambda: id(ble_client_tion).set_enabled(false);

wifi:
  ssid: "<SNIP>"
  password: "<SNIP>"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tion Fallback Hotspot"
    password: "<SNIP>"

esphome:
  name: tion

esp32_ble_tracker:

ble_client:
  - mac_address: $mac_tion
    id: ble_client_tion

button:
  - platform: template
    name: "$name Pair"
    on_press:
      lambda: id(tion_climate).pair();
    entity_category: config
  - platform: template
    name: "$name Reset Pair"
    on_press:
      lambda: id(tion_climate).reset_pair();
    entity_category: config

climate:
  - platform: tion_3s
    id: tion_climate
    ble_client_id: ble_client_tion
    name: "$name"
    buzzer:
      name: "$name Buzzer"
    outdoor_temperature:
      name: "$name Temp (Outside)"
    filter_days_left:
      name: "$name Filter Days Left"
    air_intake:
      name: "$name Air Intake"
    update_interval: 60s

sensor:
  - platform: wifi_signal
    name: "$name Wi-Fi Signal"
    update_interval: 60s
```

</p>
</details>

Получается, что сейчас всё работает отлично, кроме двух моментов:
* Перепутаны температуры входящего (уличного) воздуха и воздуха на выходе из бризера (после подогрева);
* Счётчик времени жизни фильтров, судя по всему, показывает время наработки фильтров (в моём случае 100 дней работы), а не оставшийся срок их службы (260 дней).

Ниже приложу скриншоты из официального приложения Tion и из HA, где видна разница в показаниях.

<details><summary>HomeAssistant</summary>
<p>
[image]
</p>
</details>

<details><summary>Tion App</summary>
<p>

[image]

</p>
</details>

dentra, 2022-05-11:
temp_in и temp_out удалены, вместо этого температура с датчика внутри помещения отображается в компоненте климата, сенсор для температуры снаружи называется outdoor_temperature, set_direct_write так же удален тк больше не требуется.

dentra, 2022-05-11:
@spijet подозреваю что с VERY_VERBOSE так звезды сошлись, к сожалению, сопряжение очень нестабильное.

esphome.on_boot можно не писать, все что надо будет прописано автоматом. 

температура после подогрева должна быть прямо в компоненте климата, внешняя температура в сенсоре outdoor_temperature. если неправильно - поправлю.

Так же прошу посмотреть в лог в режиме  VERBOSE, компонент после запроса состояния будет выводить такие строки:
```
[V][tion_3s:101]: fan_speed    : 3
[V][tion_3s:102]: gate_position: 2
[V][tion_3s:103]: target_temp  : 12
[V][tion_3s:104]: heater_state : ON
[V][tion_3s:105]: power_state  : ON
[V][tion_3s:106]: timer_state  : OFF
[V][tion_3s:107]: sound_state  : OFF
[V][tion_3s:108]: auto_state   : OFF
[V][tion_3s:109]: ma_connect   : OFF
[V][tion_3s:110]: save         : OFF
[V][tion_3s:111]: ma_pairing   : OFF
[V][tion_3s:112]: reserved     : 0x00
[V][tion_3s:113]: outdoor_temp1: 15
[V][tion_3s:114]: outdoor_temp2: 15
[V][tion_3s:115]: indoor_temp  : 14
[V][tion_3s:116]: filter_time  : 65322
[V][tion_3s:117]: hours        : 16
[V][tion_3s:118]: minutes      : 59
[V][tion_3s:119]: last_error   : 0
[V][tion_3s:120]: productivity : 60
[V][tion_3s:121]: filter_days  : 476
[V][tion_3s:122]: firmware     : 0032
```

посмотрите пожалуйста, что не соответствует.

spijet, 2022-05-11:
Статистика из лога сейчас выглядит так:
```
[01:16:18][V][tion_3s:104]: fan_speed    : 4
[01:16:18][V][tion_3s:105]: gate_position: 2
[01:16:18][V][tion_3s:106]: target_temp  : 19
[01:16:18][V][tion_3s:107]: heater_state : ON
[01:16:18][V][tion_3s:108]: power_state  : ON
[01:16:18][V][tion_3s:109]: timer_state  : OFF
[01:16:18][V][tion_3s:110]: sound_state  : ON
[01:16:18][V][tion_3s:111]: auto_state   : OFF
[01:16:18][V][tion_3s:112]: ma_connect   : OFF
[01:16:18][V][tion_3s:113]: save         : OFF
[01:16:18][V][tion_3s:114]: ma_pairing   : OFF
[01:16:18][V][tion_3s:122]: last_error   : 0
[01:16:18][V][tion_3s:123]: productivity : 75
[01:16:18][V][tion_3s:124]: filter_days  : 103
[01:16:18][V][tion_3s:125]: firmware     : 003C
```

Заметил, что некоторых данных нет в выводе вообще. Возможно, сейчас из-за плохого сигнала (унёс ESP32 в другую комнату, теперь до бризера ~5 метров и гипсокартоновая стена с профилем 75мм). :) Перенесу поближе и попробую ещё раз.

spijet, 2022-05-11:
Вернул лог в `VERBOSE`, перенёс ESP32 чуть ближе, получил такой вывод:
```
[01:50:14][V][tion_3s:104]: fan_speed    : 5
[01:50:14][V][tion_3s:105]: gate_position: 2
[01:50:14][V][tion_3s:106]: target_temp  : 19
[01:50:14][V][tion_3s:107]: heater_state : ON
[01:50:14][V][tion_3s:108]: power_state  : ON
[01:50:14][V][tion_3s:109]: timer_state  : OFF
[01:50:14][V][tion_3s:110]: sound_state  : ON
[01:50:14][V][tion_3s:115]: reserved     : 0x00
[01:50:14][V][tion_3s:116]: outdoor_temp1: 19
[01:50:14][V][tion_3s:117]: outdoor_temp2: 21
[01:50:14][V][tion_3s:118]: indoor_temp  : 8
[01:50:14][V][tion_3s:119]: filter_time  : 257
[01:50:14][V][tion_3s:120]: hours        : 1
[01:50:14][V][tion_3s:121]: minutes      : 45
[01:50:14][V][tion_3s:122]: last_error   : 0
[01:50:14][V][tion_3s:123]: productivity : 90
[01:50:14][V][tion_3s:124]: filter_days  : 103
[01:50:14][V][tion_3s:125]: firmware     : 003C
```

Собсно, что сейчас могу сказать:
* `target_temp` — настроенная температура подогрева (показывает правильно, 19 градусов);
* `outdoor_temp1` — не уверен, возможно дублируется настроенная целевая?
* `outdoor_temp2` — текущая температура воздуха *после нагревателя* (т.е. текущая температура внутри помещения);
* `indoor_temp` — температура воздуха *на входе в бризер* (т.е. текущая температура на улице);
* `filter_time` — остаточный ресурс фильтров в днях (то, что показывается в приложении как "время жизни фильтров");
* `filter_days` — обратная предудущей метрика (т.е. сколько дней прошло с момента установки новых фильтров);
* `productivity` — текущая производительность бризера в кубометрах в час (можно вынести её в опциональный сенсор в HA, если вдруг кому надо);
* `gate_position` — положение заслонки на входе (в моём случае `2` значит "забор воздуха с улицы");
* `{power,heater,sound,timer}_state` похожи на правду.

Получается, нужно в сенсор срока жизни фильтров отдавать число из `filter_time` (может переименовать в `filter_remaining`?), а `indoor_temp` и `outdoor_temp2` поменять местами.
Ещё заметил, что настройка `Tion 3S Air Intake` в HA всё время висит в состоянии "Unknown" (см. скрин ниже). Возможно, значения у 3S и 4S/Lite отличаются, и поэтому он не может понять, что значит `2` в этом поле?
[image]

dentra, 2022-05-17:
@spijet Спасибо за детализацию, в ближайшее время внесу изменения

spijet, 2022-05-25:
Потестировал этот модуль последнюю неделю, кроме вышеописанного нашёл только одну проблему — примерно через сутки-двое работы BLE подвисает и перестаёт обмениваться данными с бризером (при этом сама ESP32 отвечает и принимает команды от HA). Причём, судя по симптомам, BLE подвисает в состоянии "коннект открыт", т.к. родной пульт бризера не может к нему подключиться (3S может держать только одно BLE-подключение одновременно). Помогает вырубить ESP по питанию, подождать пару минут и включить обратно.

ЕМНИП, авторы [TionAPI](https://github.com/TionAPI) и [go-tion](https://github.com/m-pavel/go-tion) используют для 3S схему работы "подключился, забрал данные, отдал команды, отключился", чтобы избежать проблем с зависанием коннекта и конфликтов с другими устройствами, вроде пультов и телефонов.

dext0r, 2022-05-25:
> Потестировал этот модуль последнюю неделю, кроме вышеописанного нашёл только одну проблему — примерно через сутки-двое работы BLE подвисает и перестаёт обмениваться данными с бризером (при этом сама [...]

У меня аналогичная проблема возникала пару раз на 4S. Не знаю совпадение ли, но вероятность её появления выше если стримить логи по сети через `esphome logs`. Помогает обычный ребут ESP через button (`platform: reboot`)
 
> ЕМНИП, авторы [TionAPI](https://github.com/TionAPI) и [go-tion](https://github.com/m-pavel/go-tion) используют для 3S схему работы "подключился, забрал данные, отдал команды, отключился", чтобы избежать проблем с зависанием коннекта и конфликтов с другими устройствами, вроде пультов и телефонов.

Здесь так же, это видно в логах.

spijet, 2022-05-25:
@dext0r, в моём случае простой ребут (через `platform: reboot`, кнопку на плате ESP или даже по питанию) не помогает, нужно именно оборвать связь на достаточно долгое время, чтобы BLE-стек бризера "развис". Если получится заставить ESP по расписанию выключать BLE на минуту-две — будет достаточно.

dentra, 2022-05-25:
У меня используется такая же схема (подключаемся, забираем данные, отключаемся), как часто подключаться настраивается через update_interval, но есть нюанс, отключение происходит только после получения ответа, если ответа нет, то все как бы зависает (нет, но выгляди так). добавлю настраиваемый таймаут чтобы принудительно отключался.

spijet, 2022-05-25:
Я сначала использовал интервал 5, потом 30 и сейчас 60 секунд, чтобы не теребить бризер слишком часто. Заметил, что сенсор с уровнем сигнала от бризера всё равно обновляется каждые 5 секунд. Изменить интервал у него не вышло, поэтому повесил на него фильтр `throttle_average`. :)

dentra, 2022-05-25:
60 секунд (по умолчанию) в самый раз. Если выполняете какую-то команду, например, меняете температуру, то она автоматом инициирует подключение. Уровень сигнала от бризера - это ble_rssi? елси так, то он обновляет сразу данные как они приходят, поэтому да может немного спамить и throttle_average идеальное решение.

spijet, 2022-05-25:
Да, он самый. :)

dentra, 2022-05-26:
* Поменял местами сенсоры температуры (3s)
* Надеюсь пофиксил работу с air_intake (3s)
* Добавил параметр state_timeout для контроля ожидания состояния. 15 секунд по умолчанию, 0 для выключения (3s,Lt,4S)

Плюс ранее добавил пресеты, пример:
```yaml
  # Awailable configurable presets: home, away, boost, comfort, eco, sleep, activity.
  # Defaults:
  #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
  #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
  #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
  #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
  #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
  #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
  #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
  # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
  presets:
    home: { fan_speed: 1, target_temperature: 15, mode: heat }
    away: { fan_speed: 1, target_temperature: 20, mode: fan_only }
    boost:
    sleep: { fan_speed: 1, target_temperature: 15, mode: heat }
  # Optional, boost time
  boost_time:
    name: $name Boost Time
  # Optional, display remaining boost time
  boost_time_left:
    name: $name Boost Time Left
 ```

spijet, 2022-05-27:
Отлично, спасибо!
Поставил новую версию, буду проверять на неделе. Уже могу подтвердить, что температуру и режим забора воздуха теперь показывает правильно. Осталось проверить таймауты и починить "Filter Days Left" (`filter_days` -> поменять его местами с `filter_time`). :)

<details><summary>Снимок из HomeAssistant</summary>
<p>

[image]

</p>
</details>

spijet, 2022-05-28:
Посреди ночи ESP перестала подключаться к бризеру, в логе сыпет вот такими ошибками 2-3 раза в секунду:
```
[13:21:40][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[13:21:40][W][ble_client:117]: connect to f2:ed:73:31:4f:7d failed, status=133
[13:21:40][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[13:21:41][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[13:21:41][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 62
[13:21:41][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[13:21:41][W][ble_client:117]: connect to f2:ed:73:31:4f:7d failed, status=133
[13:21:41][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[13:21:41][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[13:21:41][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 62
[13:21:41][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[13:21:41][W][ble_client:117]: connect to f2:ed:73:31:4f:7d failed, status=133
[13:21:42][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[13:21:42][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[13:21:42][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 62
[13:21:42][V][ble_client:115]: [f2:ed:73:31:4f:7d] ESP_GATTC_OPEN_EVT
[13:21:42][W][ble_client:117]: connect to f2:ed:73:31:4f:7d failed, status=133
[13:21:42][D][ble_client:047]: Found device at MAC address [F2:ED:73:31:4F:7D]
[13:21:42][I][ble_client:085]: Attempting BLE connection to f2:ed:73:31:4f:7d
[13:21:42][V][ble_client:147]: [f2:ed:73:31:4f:7d] ESP_GATTC_DISCONNECT_EVT, reason 62
```

dentra, 2022-06-06:
Пофиксил отображение `Filter Days Left`
Добавил экспериментальную опцию `exterimental_always_pair`, если установить ее в `true`, то при подключении всегда будет посылаться дополнительная команда активации (есть репорты, что бризер перестает отдавать состояние)

По отвалу соединения здравых мыслей к сожалению пока нет, по логу esp просто не может соедениться с бризером, до компонента управление даже не доходит :(

spijet, 2022-06-07:
@dentra, собрал вчера на ночь новый мастер — пока всё отлично, обрывов или зависаний с `exterimental_always_pair: true` за ~9 часов не было. Продолжаю наблюдение. Время жизни фильтров теперь показывает правильно, спасибо! :)

spijet, 2022-06-09:
Рано обрадовался — через два дня повторилась проблема с дисконнектом со `status=133`. :|
Попробую поребутать ESP и поиграться с таймаутами.

spijet, 2022-06-09:
Наверное да, может при ожидании в ~10-30 секунд при ошибке оно будет успевать продышаться.

spijet, 2022-07-11:
За этот месяц случилось забавное. Со времени последнего комментария я примерно неделю пытался заставить ESP'шку снова подружиться с бризером и не смог — не помогли ни пляски с отключением ESP от питания, ни с выключением бризера. Решил отложить это всё на попозже, но оставил ESP включенной.
В один прекрасный день началась гроза и скакнуло напряжение — сработала защита на вводе в квартиру и отрубила питание на 10 секунд. После этого ESP запустилась и сама без каких-либо ухищрений подключилась к бризеру и проработала без ошибок 3 недели! Но потом проблема всё-таки повторилась. :)

dentra, 2022-07-14:
@spijet возможно можно попробовать отключать питание бризера, может он чего накапливает? Для переподключения говорят помогает "тыкнуть" родной пульт. Так же попробуйте использовать `persistent_connection: true`

dentra, 2022-07-14:
- пофиксил ошибку с переключением потока воздуха
- сделал воркараунд для возможности сразу включить режим подогрева из режима вентиляции без выключения (через выключение, но автоматом)

dentra, 2022-07-14:
завел группу в телеграм https://github.com/dentra/esphome-tion присоединяйтесь

dentra, 2022-07-15:
Ошибся со ссылкой, вот правильная https://t.me/esphome_tion

dentra, 2022-07-21:
* добавил сенсор с текущей производительностью бризера в кубометрах в час

dentra, 2022-07-22:
зарелизил 2022.7.0, ошибки и доработки прошу открывать отдельными задачам

closed 2022-07-22 by dentra