dentra/esphome-tion #2: Не работает с TionLite

g84, 2022-02-18:
Прошил плату [esp32-wroom-32u](https://aliexpress.ru/item/32864722159.html) традиционным [способом](https://esphome.io/)

Бризер не управляется, в логах сообщение
`Current Temperature: nan°C`

Файл конфигурации esp32-20220218.yaml
```
esphome:
  name: esp32-20220218

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:
  password: "3b228ef79d1c0ab6150f91f149482137"
  on_begin:
    then:
      - lambda: id(ble_client_tion).set_enabled(false);

wifi:
  ssid: home
  password: password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp32-20220218"
    password: "KQvv1gbGWqng"

captive_portal:

external_components:
  - source: github://dentra/esphome-tion@2022.2.0

esp32_ble_tracker:

ble_client:
  - mac_address: F8:E3:9B:FB:FD:3E
    id: ble_client_tion

climate:
  - platform: tion_lt
    ble_client_id: ble_client_tion
    name: "Tion Lite"
    buzzer:
      name: "Tion Lite Buzzer"
    led:
      name: "Tion Lite Led"
    temp_in:
      name: "Tion Lite Temp in"
    temp_out:
      name: "Tion Lite Temp out"
    heater_power:
      name: "Tion Lite Heater power"
    airflow_counter:
      name: "Tion Lite Airflow counter"
    filter_days_left:
      name: "Tion Lite Filter Days Left"
    filter_warnout:
      name: "Tion Lite Filter Warnout"
    version:
      name: "Tion Lite Version"
```

Сообщения журнала
```
INFO Reading configuration /config/esp32-20220218.yaml...
INFO Starting log output from esp32-20220218.local using esphome API
INFO Successfully connected to esp32-20220218.local
[19:43:39][I][app:102]: ESPHome version 2022.2.3 compiled on Feb 18 2022, 19:37:53
[19:43:39][C][wifi:491]: WiFi:
[19:43:39][C][wifi:353]:   Local MAC: 08:3A:F2:7D:4B:D4
[19:43:39][C][wifi:354]:   SSID: [redacted]
[19:43:39][C][wifi:355]:   IP Address: 192.168.1.11
[19:43:39][C][wifi:357]:   BSSID: [redacted]
[19:43:39][C][wifi:358]:   Hostname: 'esp32-20220218'
[19:43:39][C][wifi:360]:   Signal strength: -88 dB ▂▄▆█
[19:43:40][C][wifi:364]:   Channel: 1
[19:43:40][C][wifi:365]:   Subnet: 255.255.255.0
[19:43:40][C][wifi:366]:   Gateway: 192.168.1.1
[19:43:40][C][logger:233]: Logger:
[19:43:40][C][logger:234]:   Level: DEBUG
[19:43:40][C][esp32_ble_tracker:709]: BLE Tracker:
[19:43:40][C][esp32_ble_tracker:710]:   Scan Duration: 300 s
[19:43:40][C][esp32_ble_tracker:711]:   Scan Interval: 320.0 ms
[19:43:40][C][esp32_ble_tracker:712]:   Scan Window: 30.0 ms
[19:43:40][C][esp32_ble_tracker:713]:   Scan Type: ACTIVE
[19:43:40][C][ble_client:035]: BLE Client:
[19:43:40][C][ble_client:036]:   Address: f8:e3:9b:fb:fd:3e
[19:43:40][C][captive_portal:144]: Captive Portal:
[19:43:40][C][mdns:084]: mDNS:
[19:43:40][C][mdns:085]:   Hostname: esp32-20220218
[19:43:40][C][ota:085]: Over-The-Air Updates:
[19:43:40][C][ota:086]:   Address: esp32-20220218.local:3232
[19:43:40][C][ota:089]:   Using Password.
[19:43:40][C][api:138]: API Server:
[19:43:40][C][api:139]:   Address: esp32-20220218.local:6053
[19:43:46][I][ota:105]: Boot seems successful, resetting boot loop counter.
[19:43:46][D][esp32.preferences:114]: Saving preferences to flash...
[19:43:47][D][esp32_ble_tracker:217]: Starting scan...
[19:43:54][D][climate:009]: 'Tion Lite' - Setting
[19:43:54][D][climate:013]:   Mode: OFF
[19:43:54][D][tion:116]: Set mode 0
[19:43:54][D][climate:381]: 'Tion Lite' - Sending state:
[19:43:54][D][climate:384]:   Mode: OFF
[19:43:54][D][climate:392]:   Custom Fan Mode: 5
[19:43:54][D][climate:404]:   Current Temperature: nan°C
[19:43:54][D][climate:410]:   Target Temperature: 0.00°C
[19:44:16][D][climate:009]: 'Tion Lite' - Setting
[19:44:16][D][climate:013]:   Mode: HEAT
[19:44:16][D][tion:116]: Set mode 3
[19:44:16][D][climate:381]: 'Tion Lite' - Sending state:
[19:44:16][D][climate:384]:   Mode: HEAT
[19:44:16][D][climate:392]:   Custom Fan Mode: 5
[19:44:16][D][climate:404]:   Current Temperature: nan°C
[19:44:16][D][climate:410]:   Target Temperature: 0.00°C
[19:44:24][D][climate:009]: 'Tion Lite' - Setting
[19:44:24][D][climate:013]:   Mode: HEAT
[19:44:24][D][tion:116]: Set mode 3
[19:44:24][D][climate:381]: 'Tion Lite' - Sending state:
[19:44:24][D][climate:384]:   Mode: HEAT
[19:44:24][D][climate:392]:   Custom Fan Mode: 5
[19:44:24][D][climate:404]:   Current Temperature: nan°C
[19:44:24][D][climate:410]:   Target Temperature: 0.00°C
```

g84, 2022-02-18:
После установки антенны на esp32-wroom-32u в логах сообщение
`[20:07:38][W][ble_client:117]: connect to f8:e3:9b:fb:fd:3e failed, status=133`

```
[20:07:38][D][esp32_ble_tracker:217]: Starting scan...
[20:07:38][D][esp32_ble_tracker:723]: Found device D3:EA:E1:BF:8B:8E RSSI=-73
[20:07:38][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:07:38][D][esp32_ble_tracker:746]:   Name: 'Tion Breezer 3S'
[20:07:38][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:07:38][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:07:38][W][ble_client:117]: connect to f8:e3:9b:fb:fd:3e failed, status=133
[20:07:38][D][esp32_ble_tracker:217]: Starting scan...
[20:07:38][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:07:38][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:07:38][W][ble_client:117]: connect to f8:e3:9b:fb:fd:3e failed, status=133
[20:07:38][D][esp32_ble_tracker:217]: Starting scan...
[20:07:38][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:07:38][D][esp32_ble_tracker:723]: Found device D3:EA:E1:BF:8B:8E RSSI=-70
[20:07:38][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:07:38][D][esp32_ble_tracker:746]:   Name: 'Tion Breezer 3S'
[20:07:38][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:07:39][W][ble_client:117]: connect to f8:e3:9b:fb:fd:3e failed, status=133
[20:07:39][D][esp32_ble_tracker:217]: Starting scan...
[20:07:39][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:07:39][D][esp32_ble_tracker:723]: Found device D3:EA:E1:BF:8B:8E RSSI=-69
[20:07:39][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:07:39][D][esp32_ble_tracker:746]:   Name: 'Tion Breezer 3S'
[20:07:39][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:07:39][D][esp32_ble_tracker:723]: Found device C8:0F:10:AA:BA:D2 RSSI=-60
[20:07:39][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[20:07:39][D][esp32_ble_tracker:746]:   Name: 'MI_SCALE'
[20:07:39][W][ble_client:117]: connect to f8:e3:9b:fb:fd:3e failed, status=133
[20:07:39][D][esp32_ble_tracker:217]: Starting scan...
[20:07:39][D][esp32_ble_tracker:723]: Found device D3:EA:E1:BF:8B:8E RSSI=-73
[20:07:39][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:07:39][D][esp32_ble_tracker:746]:   Name: 'Tion Breezer 3S'
```

dentra, 2022-02-18:
а Вы вводили бризер в режим сопряжения? одни раз, как и мобильным приложением, это необходимо сделать.

g84, 2022-02-18:
Да, ввел в режим сопряжения, теперь ок. Сапасибо.

```
[20:25:14][D][esp32_ble_tracker:217]: Starting scan...
[20:25:15][D][esp32_ble_tracker:723]: Found device D3:EA:E1:BF:8B:8E RSSI=-70
[20:25:15][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:25:15][D][esp32_ble_tracker:746]:   Name: 'Tion Breezer 3S'
[20:25:15][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:25:15][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:25:15][D][tion:078]: Pairing complete
[20:25:15][I][ble_client:161]: Service UUID: 0x1800
[20:25:15][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[20:25:15][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0xa
[20:25:15][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[20:25:15][I][ble_client:347]:  characteristic 0x2A04, handle 0x7, properties 0x2
[20:25:16][I][ble_client:347]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[20:25:16][I][ble_client:161]: Service UUID: 0x1801
[20:25:16][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xd
[20:25:16][I][ble_client:347]:  characteristic 0x2A05, handle 0xc, properties 0x20
[20:25:16][I][ble_client:161]: Service UUID: 98F00001-3788-83EA-453E-F52244709DDB
[20:25:16][I][ble_client:162]:   start_handle: 0xe  end_handle: 0xffff
[20:25:16][I][ble_client:347]:  characteristic 98F00002-3788-83EA-453E-F52244709DDB, handle 0x10, properties 0xc
[20:25:16][D][tion:070]: Discovering complete
[20:25:16][D][esp32_ble_tracker:217]: Starting scan...
[20:25:16][D][text_sensor:067]: 'Tion Lite Version': Sending state '054B'
[20:25:16][I][tion:165]: Work Mode       : 01
[20:25:16][I][tion:166]: Device type     : 8002
[20:25:16][I][tion:167]: Device sub-type : 0000
[20:25:16][I][tion:168]: Hardware version: 0001
[20:25:16][I][tion:169]: Firmware version: 054B
[20:25:16][D][tion-api-lt:119]: Request state
[20:25:16][D][tion-api-lt:124]: Write state
[20:25:16][D][switch:037]: 'Tion Lite Led': Sending state OFF
[20:25:16][D][sensor:125]: 'Tion Lite Temp in': Sending state 6.00000 °C with 0 decimals of accuracy
[20:25:16][D][sensor:125]: 'Tion Lite Temp out': Sending state 21.00000 °C with 0 decimals of accuracy
[20:25:16][D][sensor:125]: 'Tion Lite Heater power': Sending state 0.00000 W with 0 decimals of accuracy
[20:25:16][D][sensor:125]: 'Tion Lite Airflow counter': Sending state 126873.52344 m³ with 2 decimals of accuracy
[20:25:16][D][binary_sensor:036]: 'Tion Lite Filter Warnout': Sending state OFF
[20:25:16][D][sensor:125]: 'Tion Lite Filter Days Left': Sending state 78.00000 days with 0 decimals of accuracy
[20:25:16][I][tion_lt:087]: last_com_source       : 1
[20:25:16][I][tion_lt:088]: auto_co2              : 0
INFO esp32-20220218.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-20220218.local
WARNING Disconnected from API
INFO esp32-20220218.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-20220218.local
[20:25:28][D][climate:009]: 'Tion Lite' - Setting
[20:25:28][D][climate:013]:   Mode: OFF
[20:25:28][D][tion:116]: Set mode 0
[20:25:28][D][climate:381]: 'Tion Lite' - Sending state:
[20:25:28][D][climate:384]:   Mode: OFF
[20:25:28][D][climate:392]:   Custom Fan Mode: 4
[20:25:28][D][climate:404]:   Current Temperature: nan°C
[20:25:28][D][climate:410]:   Target Temperature: 7.00°C
[20:25:28][D][ble_client:047]: Found device at MAC address [F8:E3:9B:FB:FD:3E]
[20:25:28][I][ble_client:085]: Attempting BLE connection to f8:e3:9b:fb:fd:3e
[20:25:28][D][tion:078]: Pairing complete
[20:25:29][I][ble_client:161]: Service UUID: 0x1800
[20:25:29][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[20:25:29][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0xa
[20:25:29][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[20:25:29][I][ble_client:347]:  characteristic 0x2A04, handle 0x7, properties 0x2
[20:25:29][I][ble_client:347]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[20:25:29][I][ble_client:161]: Service UUID: 0x1801
[20:25:29][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xd
[20:25:29][I][ble_client:347]:  characteristic 0x2A05, handle 0xc, properties 0x20
[20:25:29][I][ble_client:161]: Service UUID: 98F00001-3788-83EA-453E-F52244709DDB
[20:25:29][I][ble_client:162]:   start_handle: 0xe  end_handle: 0xffff
[20:25:29][I][ble_client:347]:  characteristic 98F00002-3788-83EA-453E-F52244709DDB, handle 0x10, properties 0xc
[20:25:29][I][ble_client:347]:  characteristic 98F00003-3788-83EA-453E-F52244709DDB, handle 0x12, properties 0x10
[... 109 lines cut ...]
[20:25:39][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:25:39][D][esp32_ble_tracker:748]:   TX Power: 2
[20:25:40][I][ble_client:075]: [f8:e3:9b:fb:fd:3e] Disabling BLE client.
[20:25:41][D][esp32_ble_tracker:723]: Found device F8:E3:9B:FB:FD:3E RSSI=-56
[20:25:41][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:25:41][D][esp32_ble_tracker:746]:   Name: 'Br Lite'
[20:25:41][D][esp32_ble_tracker:723]: Found device 6D:19:71:EB:A4:72 RSSI=-84
[20:25:41][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:25:41][D][esp32_ble_tracker:748]:   TX Power: 2
[20:25:43][D][esp32_ble_tracker:723]: Found device EC:FA:5C:7E:88:08 RSSI=-74
[20:25:43][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[... cut ...]
[20:25:58][D][esp32_ble_tracker:723]: Found device 28:39:5E:DE:CD:34 RSSI=-97
[20:25:58][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[20:26:00][D][esp32_ble_tracker:723]: Found device C8:0F:10:AA:BA:D2 RSSI=-66
[20:26:00][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[20:26:00][D][esp32_ble_tracker:746]:   Name: 'MI_SCALE'
[20:26:01][D][esp32_ble_tracker:723]: Found device 71:FA:C7:3D:37:DF RSSI=-79
[20:26:01][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[20:26:01][D][esp32_ble_tracker:748]:   TX Power: 2
```

dentra, 2022-02-18:
отлично!