dentra/esphome-tion #3: Постоянное подключение к бризеру

dext0r, 2022-05-28:
На Tion 4S испытываю периодический отвал бризера с невозможностью дальнейшего подключения. Иногда помогает перезагрузка ESP, иногда только полная перезагрузка бризера.

Возможно имеет смысл рассмотреть опциональную возможность постоянного соединения с бризером? Моим бризером управляет только HA и подключение других клиентов не требуется.

У себя попробовал реализовать это патчем ниже, посмотрим на стабильность работы в течение нескольких дней

```diff
--- a/components/tion_4s/tion_4s.cpp
+++ b/components/tion_4s/tion_4s.cpp
@@ -26,7 +26,6 @@ void Tion4s::read(const tion_dev_status_t &status) {
   // this->request_timers();

   this->request_state();
-  this->schedule_disconnect(this->state_timeout_);
 };

 void Tion4s::read(const tion4s_time_t &tion_time) {
@@ -161,9 +160,6 @@ void Tion4s::read(const tion4s_state_t &state) {
   ESP_LOGV(TAG, "work_time      : %u", state.counters.work_time);
   ESP_LOGV(TAG, "fan_time       : %u", state.counters.fan_time);
   ESP_LOGV(TAG, "errors         : %u", state.errors);
-
-  // leave 3 sec connection left to end all of jobs
-  this->schedule_disconnect();
 }

 void Tion4s::flush_state_(const tion4s_state_t &state_) const {
diff --git a/components/tion_4s/tion_4s.h b/components/tion_4s/tion_4s.h
index 2740766..9921d23 100644
--- a/components/tion_4s/tion_4s.h
+++ b/components/tion_4s/tion_4s.h
@@ -22,7 +22,13 @@ class Tion4s : public TionClimateComponent, public Tion<TionApi4s>,public TionDi
     return traits;
   }

-  void update() override { this->parent_->set_enabled(true); }
+  void update() override {
+    if (this->node_state == esp32_ble_tracker::ClientState::ESTABLISHED) {
+      this->on_ready();
+    } else {
+      this->parent_->set_enabled(true);
+    }
+  }

   void on_ready() override;

@@ -31,7 +37,11 @@ class Tion4s : public TionClimateComponent, public Tion<TionApi4s>,public TionDi
   bool write_state() override {
     this->publish_state();
     this->update_flag_ |= UPDATE_STATE;
-    this->parent_->set_enabled(true);
+    if (this->node_state == esp32_ble_tracker::ClientState::ESTABLISHED) {
+      this->on_ready();
+    } else {
+      this->parent_->set_enabled(true);
+    }
     return true;
   };
```

dext0r, 2022-06-05:
Спустя неделю проблем не зафиксировано. Управление работает стабильно, соединение не подвисает, ESP ни разу не перезагружалась.

dentra, 2022-06-06:
Приветствую, у меня есть в todo постоянное подключение, поставлю приоритет повыше

dentra, 2022-06-06:
спасибо, за тест

dentra, 2022-06-06:
вопрос, а сам бризер не пушит состояние, надо именно полить?

dext0r, 2022-06-07:
> вопрос, а сам бризер не пушит состояние, надо именно полить?

Надо полить, без отправки фрейма `FRAME_TYPE_STATE_REQ` состояние само по себе не прилетает.

dentra, 2022-06-07:
добавил, в мастере. включается/выключается атрибутом `persistent_connection`, по у молчанию выключен (false)

dext0r, 2022-06-20:
За последние две недели проблем не замечено, всё работает как часы. Ишью закрываю. Огромное спасибо за проект!

closed 2022-06-20 by dext0r