dentra/esphome-tion #4: Tion O2 Mac

didjeru, 2022-06-17:
Привет! А есть ли вероятность, что заработает с О2 МАС? Вроде как где-то слышал, что он по WiFi работает...

dentra, 2022-06-18:
У O2 нет BLE, только если пробовать хардварное подключение

dentra, 2022-06-19:
Ещё немного информации.

Эта модель есть с дополнительным модулем на STM32+Si4463.
Общение с MA идёт по RF, частота пока неизвестна (433/470 или 868/915)

dentra, 2022-06-20:
По хардварной части все довольно просто, есть 4 пиновый разъём с 5В питанием и большой шанс что именно rx и tx, т.к. модуль общения с MA подключатся через него

rbalaev, 2023-09-01:
Есть способы подключения О2?

dentra, 2023-09-03:
К сожалению, у меня нет этого устройства. Пока кто-то не попробует удостовериться, что там есть рабочий uart и он что-то шлет и принимает никуда двинутся не могу

rbalaev, 2023-09-03:
> К сожалению, у меня нет этого устройства. Пока кто-то не попробует удостовериться, что там есть рабочий uart и он что-то шлет и принимает никуда двинутся не могу

Я могу как то вам помочь в этом, девайсы есть, но нужна помощь.

didjeru, 2023-09-03:
Добрый день! Да, у него уарт. Скорее всего у всех моделей (не только у O2 MAC, который у меня), так как модуль радиоканала можно купить отдельно. Нужно открыть и проверить наличие разъема. Для исследования, мне нужна была станция. Ну а после того как я ее приобрел, разбирать прибор больше не захотел.  Скорее всего можно попробовать считать данные и со станции. Хочется отвязаться о аккаунта в Тионе. Но одному заниматься не интересно…

rbalaev, 2023-09-04:
У меня есть RF модуль из O2. если это поможет
[image]
[image]

rbalaev, 2023-09-08:
O2 еще может работать через wifi модуль MagicAir, через такой способ можно добавить?

dentra, 2023-09-08:
Через MA какие-то данные можно получить, для этого есть компонент в HA, но он забирает данные из облака

dentra, 2023-09-08:
Спасибо за отличные фото (видно все дорожки), я посмотрю uart это или разъем для прошивки.

dentra, 2023-09-08:
какими разъемом этот модуль подключается к бризеру, X4, вот [сюда](https://github.com/dentra/esphome-tion/blob/master/hardware/o2/photo_2022-06-21_15-53-32.jpg)?

dentra, 2023-09-08:
если разъем X4 действительно юарт, то проще всего подключить RF модуль к компу через переходник и посмотреть какие данные он посылает. В целом есть полная информация о двух тионовских протоколах, если это какой-то один из них, то сразу бинго, в противном случае реверс займет чуть больше времени и потребутеся создать прокси подключение.

didjeru, 2023-09-09:
> O2 еще может работать через wifi модуль MagicAir, через такой способ можно добавить?

Нет у О2 вафли. С МА они общаются по радиоканалу. А уже МА с роутером.

dentra, 2024-01-25:
Следующая информация для тех кто хоть немного дружит с паяльником, имеет в распоряжении станцию MagicAir (MA) и имеет желание помочь собрать данные чтобы разобрать протокол взаимодействия.

MA требуется для того чтобы посылать команды управления. 
На текущем этапе подключать к esp необходимо и бризер, и RF-модуль. ESP будет проксировать и логировать все команды от MA.

Оптимальнее всего если заведется на какой-нибудь esp32c3 т.к. дешева и не требовательна по питанию.

Для подключения, нас интересует разъем X4:
[image]

Конфиг для прошивки предоставлю по вашей готовности.

Для сбора данных от вас требуется, выполнять следующие шаги:

1. знать текущее состояние бризера (чем больше данных, тем лучше)
2. текущее состояние лога бризера
3. выполнить какую-то команду
4. дождаться результата в логе
5. сохранить результаты
6. повторять по отдельности пункты 1-5

dentra, 2024-01-25:
Структура команд `XX YY..YY ZZ`, где
* `XX` - код команды, запрос всегда `0X`, ответ `1X`, но есть исключение
* `YY..YY` тело команды, может отсутствовать
* `ZZ` контрольная сумма (polynominal=0x01, initial=0xFF, final=0x00)

> В пакете с командой отсутсвует опозновательный байт, поэтому пока теоретически могут наблюдаться проблемы, когда и если появится информация о таймингах, это дело стабилизирую. У меня нет этого бризера, тестировать соответсвенно тоже не на чем, все делаю исключительно по информации, которую вы мне присылаете, поэтому прошу отнестись с терпением и пониманием.

На текущий момент известны команды:
### `00` connect (предположительно)
```
RF: 00 FF
O2: 10 04 10 01 00 FA
```
структура (тело) ответа неизвестна

команда посылается при появлении питания на RF модуле, следом сразу идет команда `07`

### `07` device status (предположительно)
```
RF: 07 F8
O2: 17 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 08 61 0E 13 04 10 EC 19 79
```
структура (тело) ответа неизвестна

### `01` state
```
RF: 01 FE
O2: 11 0С 14 17 10 02 3С 04 00 00 04 0А ВС 01 47 73 СВ 01 E3
```

структура (тело) ответа:
```
00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18
```
* 02 - int8 outdoor_temperature (предположительно)
* 04 - int8 current_temperature (предположительно)

### `03` heartbeat
```
RF: 03 FC
O2: 13 00 EC
```
структура (тело) ответа неизвестна

следом сразу идет команда `04`

### `04` work mode (предположительно)
```
RF: 04 00 FB
O2: 55 AA
```
структура (тело) запроса неизвестна, возможно `00` из ответа на запрос `03`

команды `03` и `04` повторяются каждые 200мс.

dentra, 2024-01-26:
для желающих попробовать [tion-o2-uart.yaml](../blob/master/tion-o2-uart.yaml)

dentra, 2024-01-27:
# Важно

Для правильной работы, как минимум, по линии RX необходимо согласование уровня. В оригинальной схеме RF модуля это сделано по минимуму через резистор номиналом 100 Ом. Рекомендуется преобразователь уровней или схема из 2 резисторов.

Гарантировать работу и дальнейшую поддержку могу только на чипах семейства esp32.

dentra, 2024-01-28:
Ну что же есть положительный отзыв, базовый функционал успешно работает.
Так же проверили, команды 03 и 04 для корректной работы посылать не обязательно.
Возможно какая та из команд влияет на определение подключён ли RF модуль или нет, но это повлияло только на то, что при попытке активировать MA написало «no RF».

dpolyakov, 2024-05-21:
Привет. 
подойдет ли o2 base версия для модификации?
Если да, то поделитесь пожалуйста схемой для распайки

dentra, 2024-05-21:
@dpolyakov только если выведен [такой](https://github.com/dentra/esphome-tion/blob/master/hardware/o2/photo_2022-06-21_15-53-32.jpg) 4-х пиновый разъем

dentra, 2024-08-13:
реализовано

closed 2024-08-13 by dentra