dentra/esphome-tion #5: UART for Tion 4S

dentra, 2022-08-12:
Tion 4S has a built in communication port that is simple implements a standard UART interface via USB pins. The difference is in inverted RX/TX pins.

Communication protocol is the same as BLE, but additionally we need to send heartbeat command (0x3931) and absence of support for Turbo mode.

dentra, 2022-08-12:
Sample configuration:
```yaml
uart:
  id: uart_bus
  tx_pin: RX
  rx_pin: TX
  baud_rate: 9600

vport:
  - platform: tion_uart
    # Optional, How often query device state. Default: 10s.
    update_interval: 10s
    # Optional, heartbeat interval. Default: 3s.
    heartbeat_interval: 5s

climate:
  - platform: tion_4s
    name: "$name"
    # Optional, Buzzer control switch.
    buzzer:
      name: "$name Buzzer"
    # Optional, Led control switch.
    led:
      name: "$name Led"
    # Optional, Recirculation control switch.
    recirculation:
      name: "$name Recirculation"
    # Optional, Temperatire of air before heater, °C (celsius).
    outdoor_temperature:
      name: "$name Outdoor Temperature"
    # Optional, Heater power, W (watt).
    heater_power:
      name: "$name Heater power"
    # Optional, Total airflow counter, m³ (cubic meters).
    airflow_counter:
      name: "$name Airflow counter"
    # Optional, Filter time left counter, days.
    filter_time_left:
      name: "$name Filter Time Left"
    # Optional, Filter warning state.
    filter_warnout:
      name: "$name Filter Warnout"
    # Optional, Tion firmware version.
    version:
      name: "$name Version"
    # Optional, Enable presets. Default: <empty>
    # Awailable configurable presets: home, away, boost, comfort, eco, sleep, activity.
    # Defaults:
    #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
    #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
    #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
    #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
    #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
    #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
    #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
    # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
    presets:
      # enable and override "home" preset
      home: { fan_speed: 2, target_temperature: 15, mode: heat }
      # enable "away" preset
      away:
      # enable "boost" preset
      boost:
      # enable and override "sleep" preset
      sleep: { fan_speed: 1, target_temperature: 15, mode: heat }
    # Optional, configure boost time
    boost_time:
      name: $name Boost Time
    # Optional, display remaining boost time
    boost_time_left:
      name: $name Boost Time Left
```

dentra, 2022-10-08:
Since 2022.10.0 this feature is fully supported.

DachnikGarik, 2022-10-31:
Можно железную часть описать на пальцах для домохозяек? Схему подключения к разъему.

dentra, 2022-11-02:
Распиновка по питанию как у обычного USB, только RX и TX поменяны местами. __Питание и логические уровни 5В__. Сразу предупреждаю, что Software UART может приводить к срабатыванию WDT, как минимум это раз в день/два у меня проявляется на стике от Rubetek (ESP8266EX). С Hardware UART долго не гонял, не собрать пока тестовый стик, а с висящими наружу проводами и железками мне дома бризер оставить не позволяют :)
Если есть возможность собрать на ESP32, это наверное идеальный вариант, т.к. при старте ESP не будет сыпать условным мусором в бризер (стартовый лог), а при инициализации RX/TX сменяться, но HW UART будет работать.

i2r, 2023-01-25:
У меня **завёлся** Hardware UART на Tion 4S с платой `Wemos s2 mini`: https://www.wemos.cc/en/latest/s2/s2_mini.html

На версии ESPHome 2023.1.0-dev (Jan 25 2023, 21:57:34)

С конфигом:

#### tioncontrol.yaml
```yaml
esphome:
  name: tioncontrol

esp32:
  board: lolin_s2_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_pass

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$name Fallback Hotspot"
    password: !secret ap_pass

substitutions:
  # name of the node.
  node_name: tioncontrol
  # name for main climate entity and prefix for all others.
  name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"
  # setup your board or leave it asis.
  board: $board
  # wifi settings.
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_pass
packages:
  ## Enable additional RSSI signal sensor. Comment next line to disable.
  # rssi_signal: github://dentra/esphome-tion/packages/rssi_signal.yaml
  ## Enable additional WiFi signal sensor. Comment next line to disable.
  wifi_signal: github://dentra/esphome-tion/packages/wifi_signal.yaml
  ## Enable additional Uptime sensor. Comment next line to disable.
  uptime: github://dentra/esphome-tion/packages/uptime.yaml
  ## Uncomment next line to enable additional restart button.
  # restart: github://dentra/esphome-tion/packages/restart.yaml
  ## Uncomment next line to enable persistent connection runtime control switch.
  # persistent: github://dentra/esphome-tion/packages/persistent.yaml
  ## Uncomment next line to enable additional connect switch useful for debugging.
  # connect: github://dentra/esphome-tion/packages/connect.yaml
  ## Uncomment next line to enable additional debug information.
  # debug: github://dentra/esphome-tion/packages/debug.yaml
  ## Uncomment next line to if you wish to compile via ESP IDF.
  idf: github://dentra/esphome-tion/packages/idf.yaml
  # Common configuration parameters. Do not comment or remove!
  # common: github://dentra/esphome-tion/packages/common.yaml
  common: !include common.yaml
uart:
  id: uart_bus
  tx_pin: 20
  rx_pin: 19
  baud_rate: 9600
vport:
  - platform: tion_uart
    # Optional, How often query device state. Default: 10s.
    update_interval: 10s
    # Optional, heartbeat interval. Default: 3s.
    heartbeat_interval: 5s
climate:
  - platform: tion_4s
    name: "$name"
    # Optional, Buzzer control switch.
    buzzer:
      name: "$name Buzzer"
    # Optional, Led control switch.
    led:
      name: "$name Led"
    # Optional, Recirculation control switch.
    recirculation:
      name: "$name Recirculation"
    # Optional, Temperatire of air before heater, °C (celsius).
    outdoor_temperature:
      name: "$name Outdoor Temperature"
    # Optional, Heater power, W (watt).
    heater_power:
      name: "$name Heater power"
    # Optional, Total airflow counter, m³ (cubic meters).
    airflow_counter:
      name: "$name Airflow counter"
    # Optional, Filter time left counter, days.
    filter_time_left:
      name: "$name Filter Time Left"
    # Optional, Filter warning state.
    filter_warnout:
      name: "$name Filter Warnout"
    # Optional, Tion firmware version.
    version:
      name: "$name Version"
    # Optional, Enable presets. Default: <empty>
    # Awailable configurable presets: home, away, boost, comfort, eco, sleep, activity.
    # Defaults:
    #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
    #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
    #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
    #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
    #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
    #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
    #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
    # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
    presets:
      # enable and override "home" preset
      home: { fan_speed: 2, target_temperature: 18, mode: heat }
      # enable "away" preset
      away: { fan_speed: 1, target_temperature: 10, mode: heat }
      # enable "boost" preset
      boost: { fan_speed: 4, target_temperature: 18, mode: heat }
      # enable and override "sleep" preset
      sleep: { fan_speed: 1, target_temperature: 18, mode: heat }
    # Optional, configure boost time
    boost_time:
      name: $name Boost Time
    # Optional, display remaining boost time
    boost_time_left:
      name: $name Boost Time Left
```

#### common.yaml
```yaml
substitutions:
  name: $node_name
  board: $board
  # framework settings: esp-idf or arduino
  framework_type: esp-idf
  node_add_mac_suffix: "false"
  tion_version: 2022.10.1
  tion_source: github://dentra/esphome-tion@$tion_version

esphome:
  name: $node_name
  name_add_mac_suffix: $node_add_mac_suffix
  project:
    name: "dentra.tion"
    version: "$tion_version ($framework_type)"

esp32:
  board: $board
  framework:
    type: $framework_type

preferences:
  flash_write_interval: 10min

external_components:
  - source: github://dentra/esphome-components
    components: [crash_info, etl]
  - source: $tion_source
```

dentra, 2023-07-03:
https://github.com/dentra/esphome-tion/releases/tag/2023.7.0

closed 2023-07-03 by dentra