dentra/esphome-tion #6: Update to ESPhome 2022.9.3

plesetsk, 2022-10-06:
During updating the node a got the error in logs. Update failed

```
Updating /config/esphome/esp32-climate.yaml
------------------------------------------------------------

INFO Reading configuration /config/esphome/esp32-climate.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-climate (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Library Manager: Installing Hash
INFO Installing Hash
Library Manager: Installing ESP8266WiFi
INFO Installing ESP8266WiFi
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- Update @ 1.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- Wire @ 1.0.1
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/tion/tion.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/uart/uart_debugger.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/uptime/uptime_sensor.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
src/esphome/components/tion/tion.cpp: In member function 'virtual void esphome::tion::TionBleNode::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/tion/tion.cpp:85:67: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
       auto err = esp_ble_gattc_register_for_notify(this->parent_->gattc_if, this->parent_->remote_bda, this->char_rx_);
                                                                   ^
src/esphome/components/tion/tion.cpp:85:92: error: 'class esphome::ble_client::BLEClient' has no member named 'remote_bda'
       auto err = esp_ble_gattc_register_for_notify(this->parent_->gattc_if, this->parent_->remote_bda, this->char_rx_);
                                                                                            ^
src/esphome/components/tion/tion.cpp: In member function 'virtual bool esphome::tion::TionBleNode::write_data(const uint8_t*, uint16_t) const':
src/esphome/components/tion/tion.cpp:127:50: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
   return esp_ble_gattc_write_char(this->parent_->gattc_if, this->parent_->conn_id, this->char_tx_, size,
                                                  ^
src/esphome/components/tion/tion.cpp:127:75: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
   return esp_ble_gattc_write_char(this->parent_->gattc_if, this->parent_->conn_id, this->char_tx_, size,
                                                                           ^
*** [/data/esp32-climate/.pioenvs/esp32-climate/src/esphome/components/tion/tion.cpp.o] Error 1
========================== [FAILED] Took 8.85 seconds ==========================
======== [ERROR] /config/esphome/esp32-climate.yaml ========

======================== [SUMMARY] ========================
  - /config/esphome/esp32-climate.yaml: FAILED
```

dentra, 2022-10-08:
Fixed. But due to new hardware support you need to reconfigure your component with vport, please follow examples.

plesetsk, 2022-10-09:
Works fine. Thanks.

closed 2022-10-09 by plesetsk