dentra/esphome-tion #7: Multiple breezers only with single esp32 board

facetheheat, 2023-01-26:
👋 Hi there,
is it possible to control multiple breezers only with single board? It seems not supported by configuration yaml? 

I've had a 4 x Tion Lite with externals CO2/Temp sensors. I would like to control they from one esp32. 
Thanks!

dentra, 2023-01-26:
Yes, it is possible but maximum is 3. This is limitation of ESP framework.

This is sample configuration:
```yaml
ble_client:
  - mac_address: $tion1_mac_address
    id: tion1_ble_client
  - mac_address: $tion2_mac_address
    id: tion2_ble_client

vport:
  - platform: tion_ble
    id: tion1_ble_vport
    ble_client_id: tion1_ble_client
    ...
  - platform: tion_ble
    id: tion1_ble_vport
    ble_client_id: tion2_ble_client
    ...

climate:
  - platform: tion_lt
    vport_id: tion1_ble_vport
    name: "Tion 1"
    ...
  - platform: tion_lt
    vport_id: tion2_ble_vport
    name: "Tion 2"
    ...
```

facetheheat, 2023-01-28:
Hi there, I'm trying to add config with 3 tion lite and got this issue:

% tion-lite.yaml
```
ble_client:
  - mac_address: DD:7C:AA:8D:99:17
    id: tion1_ble_client
  - mac_address: CO:F4:AO:DD:60:44
    id: tion2_ble_client
  - mac_address: FC:DA:40:F6:4B:CA
    id: tion4_ble_client
```

% esphome run tion-lite.yaml
```
INFO Reading configuration tion-lite.yaml...
WARNING Found '$tion_mac_address' (see sensor->2->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see sensor->2->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
Failed config

sensor.ble_rssi: [source .esphome/packages/6525b20a/packages/rssi_signal.yaml:2]
  platform: ble_rssi
  
  MAC Address must consist of 6 : (colon) separated parts.
  mac_address: $tion_mac_address
  name: tion-controller RSSI
  filters: 
    throttle_average: 60s
  entity_category: diagnostic
ble_client: [source tion-lite.yaml:47]
  - mac_address: DD:7C:AA:8D:99:17
    id: tion1_ble_client
  - 
    MAC Address parts must be hexadecimal values from 00 to FF.
    mac_address: CO:F4:AO:DD:60:44
    id: tion2_ble_client
  - mac_address: FC:DA:40:F6:4B:CA
    id: tion4_ble_client

```

facetheheat, 2023-01-28:
If I switch only to 1 breezer, but changed config to custom id in ble_client:

```
ble_client:
  - mac_address: $tion_mac_address
    id: tion1_ble_client

vport:
  - platform: tion_ble
    id: tion1_ble_vport
    ble_client_id: tion1_ble_client
```

I've got an error:
% esphome run tion-lite.yaml
```
INFO Reading configuration tion-lite.yaml...
Failed config

ota: [source .esphome/packages/6525b20a/packages/common.yaml:32]
  on_begin: 
    - then: 
        - 
          Couldn't find ID 'tion_ble_client'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "tion1_ble_client", "tion1_ble_vport".
          lambda: !lambda |-
            id(tion_ble_client).set_enabled(false);
  safe_mode: True
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10

```

dentra, 2023-02-13:
Please provide your config, I will edit it for you.

facetheheat, 2023-02-14:
**$ cat tion-lite-all.yaml**
```
substitutions:
  # tion mac address.
  tion1_mac_address: DD:7C:AA:8D:99:17
  tion2_mac_address: CO:F4:AO:DD:60:44

  # name of the node.
  node_name: esp-tion-lite
  # name for main climate entity and prefix for all others.
  name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"
  # setup your board or leave it asis.
  board: esp32dev
  # wifi settings.
  wifi_ssid: "MyWiFI"
  wifi_password: "MyPassPass"

packages:
  ## Enable additional RSSI signal sensor. Comment next line to disable.
  rssi_signal: github://dentra/esphome-tion/packages/rssi_signal.yaml
  ## Enable additional WiFi signal sensor. Comment next line to disable.
  wifi_signal: github://dentra/esphome-tion/packages/wifi_signal.yaml
  ## Enable additional Uptime sensor. Comment next line to disable.
  uptime: github://dentra/esphome-tion/packages/uptime.yaml
  ## Uncomment next line to enable additional restart button.
  # restart: github://dentra/esphome-tion/packages/restart.yaml
  ## Uncomment next line to enable persistent connection runtime control switch.
  # persistent: github://dentra/esphome-tion/packages/persistent.yaml
  ## Uncomment next line to enable additional connect switch useful for debugging.
  # connect: github://dentra/esphome-tion/packages/connect.yaml
  ## Uncomment next line to enable additional debug information.
  # debug: github://dentra/esphome-tion/packages/debug.yaml
  ## Uncomment next line to if you wish to compile via ESP IDF.
  # idf: github://dentra/esphome-tion/packages/idf.yaml
  # Common configuration parameters. Do not comment or remove!
  common: github://dentra/esphome-tion/packages/common.yaml

# WiFi hotspot configuration.
# Comment next 3 lines to enable manual WiFi conviguration via access point.
wifi:
  ssid: $wifi_ssid
  password: $wifi_password

ble_client:
  - mac_address: $tion1_mac_address
    id: tion1_ble_client
  - mac_address: $tion2_mac_address
    id: tion2_ble_client

# Virtual port configuration
vport:
  - platform: tion_ble
    id: tion1_ble_vport
    ble_client_id: tion1_ble_client
    # Optional, How often query device state. Default: 60s.
    update_interval: 60s
    # Optional, Do not disconnect after receiving state. Default: false.
    persistent_connection: false
    # Optional, Timeout to reconnect if no state answer received. Default: 15s.
    state_timeout: 15s
    # Optional, Allow to disable other BLE device scanning when breezer is already connected. Default: false.
    disable_scan: false

  - platform: tion_ble
    id: tion2_ble_vport
    ble_client_id: tion2_ble_client
    # Optional, How often query device state. Default: 60s.
    update_interval: 60s
[... 120 lines cut ...]

```
INFO Reading configuration tion-lite-all.yaml...
WARNING Found '$tion_mac_address' (see sensor->2->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see sensor->2->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
Failed config

sensor.ble_rssi: [source .esphome/packages/6525b20a/packages/rssi_signal.yaml:2]
  platform: ble_rssi
  
  MAC Address must consist of 6 : (colon) separated parts.
  mac_address: $tion_mac_address
  name: esp-tion-lite RSSI
  filters: 
    throttle_average: 60s
  entity_category: diagnostic
ble_client: [source tion-lite-all.yaml:45]
  - mac_address: DD:7C:AA:8D:99:17
    id: tion1_ble_client
  - 
    MAC Address parts must be hexadecimal values from 00 to FF.
    mac_address: CO:F4:AO:DD:60:44
    id: tion2_ble_client

```

dentra, 2023-02-28:
Sorry for a long delay for answer:
```yaml
substitutions:
  # tion mac address.
  tion1_mac_address: DD:7C:AA:8D:99:17
  tion2_mac_address: C0:F4:A0:DD:60:44

  # name of the node.
  node_name: esp-tion-lite
  # name for main climate entity and prefix for all others.
  name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"
  # setup your board or leave it asis.
  board: esp32dev
  # wifi settings.
  wifi_ssid: "MyWiFI"
  wifi_password: "MyPassPass"

  # framework settings: esp-idf or arduino
  framework_type: arduino

  tion_version: 2022.10.1

esphome:
  name: $node_name
  name_add_mac_suffix: $node_add_mac_suffix
  project:
    name: "dentra.tion"
    version: "$tion_version ($framework_type)"
  min_version: 2022.12.8

esp32:
  board: $board
  framework:
    type: $framework_type

preferences:
  flash_write_interval: 10min

# WiFi hotspot configuration.
wifi:
  ssid: $wifi_ssid
  password: $wifi_password
  ap:
    ssid: "$name Fallback Hotspot"

api:
ota:
logger:

esp32_ble_tracker:

external_components:
  - source: github://dentra/esphome-components@2023.2.0
    components: [crash_info, etl, otax, vport, esp32_usb_dis]
  - source: github://dentra/esphome-tion@$tion_version

ble_client:
  - mac_address: $tion1_mac_address
    id: tion1_ble_client
  - mac_address: $tion2_mac_address
    id: tion2_ble_client

# Virtual port configuration
vport:
  - platform: tion_ble
    id: tion1_ble_vport
    ble_client_id: tion1_ble_client
    # Optional, How often query device state. Default: 60s.
[... 105 lines cut ...]
    # Defaults:
    #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
    #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
    #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
    #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
    #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
    #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
    #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
    # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
    presets:
      # enable and override "home" preset
      home: { fan_speed: 2, target_temperature: 16, mode: heat }
      # enable "away" preset
      away:
      # enable "boost" preset
      boost:
      # enable and override "sleep" preset
      sleep: { fan_speed: 1, target_temperature: 18, mode: heat }
    # Optional, configure boost time
    boost_time:
      name: $name Boost Time
    # Optional, display remaining boost time
    boost_time_left:
      name: $name Boost Time Left
```

nmusatov, 2023-03-30:
```
substitutions:
  # tion mac address.
  tion1_mac_address: E1:3B:67:5D:D6:45
  tion2_mac_address: F6:ED:29:9D:03:68

  # name of the node.
  node_name: esp-tion-lite-atom
  # name for main climate entity and prefix for all others.
  name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"
  # setup your board or leave it asis.
  board: m5stack-atom
  # wifi settings.
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

  # framework settings: esp-idf or arduino
  framework_type: esp-idf

  tion_version: 2022.10.1

esphome:
  name: $node_name
  name_add_mac_suffix: $node_add_mac_suffix
  project:
    name: "dentra.tion"
    version: "$tion_version ($framework_type)"
  min_version: 2022.12.8

esp32:
  board: $board
  framework:
    type: $framework_type

preferences:
  flash_write_interval: 10min

# WiFi hotspot configuration.
wifi:
  ssid: $wifi_ssid
  password: $wifi_password
  manual_ip:
    static_ip: 192.168.249.42
    gateway: 192.168.249.1
    subnet: 255.255.255.0
    dns1: 192.168.249.1
    dns2: 1.1.1.1

ota:
logger:

esp32_ble_tracker:

external_components:
  - source: github://dentra/esphome-components@2023.2.0
    components: [crash_info, etl, otax, vport, esp32_usb_dis]
  - source: github://dentra/esphome-tion@$tion_version

ble_client:
  - mac_address: $tion1_mac_address
    id: tion1_ble_client
  - mac_address: $tion2_mac_address
    id: tion2_ble_client

# Virtual port configuration
vport:
  - platform: tion_ble
    id: tion1_ble_vport
[... 107 lines cut ...]
    # Defaults:
    #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
    #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
    #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
    #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
    #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
    #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
    #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
    # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
    presets:
      # enable and override "home" preset
      home: { fan_speed: 2, target_temperature: 16, mode: heat }
      # enable "away" preset
      away:
      # enable "boost" preset
      boost:
      # enable and override "sleep" preset
      sleep: { fan_speed: 1, target_temperature: 18, mode: heat }
    # Optional, configure boost time
    boost_time:
      name: $name Boost Time
    # Optional, display remaining boost time
    boost_time_left:
      name: $name Boost Time Left
```

facetheheat, 2023-03-30:
> @facetheheat, have you successfully used provided config ? 
Hi Nikolay, I ended up choosing separate devices for each breezer. For me it would be more convenient to have 1 esp for 1 breezer.

standby-cappuccino, 2024-10-08:
I'am trying to add 3 breezers but I cannot validate .yaml
Here is my yaml:
[3_breezers.txt](https://github.com/user-attachments/files/17296405/3_breezers.txt)

During valiation, I'm getting this error:
```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/tionlite.yaml...
WARNING Found '$tion_mac_address' (see sensor->1->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see ble_client->0->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see sensor->1->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see ble_client->0->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
Failed config

sensor.ble_rssi: [source /config/.esphome/packages/87dfe8f6/packages/rssi_signal.yaml:2]
  platform: ble_rssi
  id: tion_ble_rssi
  
  MAC Address must consist of 6 : (colon) separated parts.
  mac_address: $tion_mac_address
  name: RSSI
  filters: 
    throttle_average: 60s
  entity_category: diagnostic
```
  
  If I comment packages/rssi_signal.yaml
  I get this:
```
  INFO ESPHome 2024.9.2
INFO Reading configuration /config/tionlite.yaml...
WARNING Found '$tion_mac_address' (see ble_client->0->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
WARNING Found '$tion_mac_address' (see ble_client->0->mac_address) which looks like a substitution, but 'tion_mac_address' was not declared
Failed config

ble_client: [source /config/.esphome/packages/87dfe8f6/packages/esp32_ble.yaml:20]
  - 
    MAC Address must consist of 6 : (colon) separated parts.
    mac_address: $tion_mac_address
    id: tion_ble_client
  - mac_address: CD:XF:88:XE:74:X6
    id: tion1_ble_client
  - mac_address: F4:X7:D2:X3:91:XF
    id: tion2_ble_client
  - mac_address: F9:1X:87:XE:97:XA
    id: tion3_ble_client
```

closed 2026-10-24 by dentra