dentra/esphome-tion #8: Проблема подключения Tion 3S

fishenebelny, 2023-01-31:
Здравствуйте,

Большое спасибо за интеграцию. Очень много проблема решает с управлением и автоматизацией бризеров.

У меня на данный момент примерно через 10 часов работы ESP32 перестает подключаться к Тиону и не получает с него информацию. В логах ESP32 следующее:
```
[09:00:44][D][esp32_ble_client:048]: [0] [D5:91:DD:FC:1A:60] Found device
[09:00:44][I][esp32_ble_client:064]: [0] [D5:91:DD:FC:1A:60] 0x01 Attempting BLE connection
[09:00:45][V][esp32_ble_client:111]: [0] [D5:91:DD:FC:1A:60] gattc_event_handler: event=41 gattc_if=4
[09:00:45][V][esp32_ble_client:164]: [0] [D5:91:DD:FC:1A:60] ESP_GATTC_DISCONNECT_EVT, reason 62
[09:00:45][V][esp32_ble_client:111]: [0] [D5:91:DD:FC:1A:60] gattc_event_handler: event=2 gattc_if=4
[09:00:45][V][esp32_ble_client:126]: [0] [D5:91:DD:FC:1A:60] ESP_GATTC_OPEN_EVT
[09:00:45][W][esp32_ble_client:131]: [0] [D5:91:DD:FC:1A:60] Connection failed, status=133
[09:00:45][D][esp32_ble_client:048]: [0] [D5:91:DD:FC:1A:60] Found device
[09:00:45][I][esp32_ble_client:064]: [0] [D5:91:DD:FC:1A:60] 0x01 Attempting BLE connection
[09:00:45][V][esp32_ble_client:111]: [0] [D5:91:DD:FC:1A:60] gattc_event_handler: event=41 gattc_if=4
[09:00:45][V][esp32_ble_client:164]: [0] [D5:91:DD:FC:1A:60] ESP_GATTC_DISCONNECT_EVT, reason 62
[09:00:45][V][esp32_ble_client:111]: [0] [D5:91:DD:FC:1A:60] gattc_event_handler: event=2 gattc_if=4
[09:00:45][V][esp32_ble_client:126]: [0] [D5:91:DD:FC:1A:60] ESP_GATTC_OPEN_EVT
[09:00:45][W][esp32_ble_client:131]: [0] [D5:91:DD:FC:1A:60] Connection failed, status=133
```
Для решения в первый раз перепрошил ESP32, а потом сделал заново сопряжение. В чем может быть проблема?
Заранее спасибо!

fishenebelny, 2023-01-31:
yaml

```
substitutions:
  # tion mac address. setup yours.
  tion_mac_address: D5:91:DD:FC:1A:60
  # name of the node.
  node_name: tion-kids
  # name for main climate entity and prefix for all others.
  name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"
  # setup your board or leave it as is.
  board: esp32dev
  # wifi settings.
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

packages:
  ## Enable additional RSSI signal sensor. Comment next line to disable.
  rssi_signal: github://dentra/esphome-tion/packages/rssi_signal.yaml
  ## Enable additional WiFi signal sensor. Comment next line to disable.
  wifi_signal: github://dentra/esphome-tion/packages/wifi_signal.yaml
  ## Enable additional Uptime sensor. Comment next line to disable.
  uptime: github://dentra/esphome-tion/packages/uptime.yaml
  ## Uncomment next line to enable additional restart button.
  restart: github://dentra/esphome-tion/packages/restart.yaml
  ## Uncomment next line to enable persistent connection runtime control switch.
  #persistent: github://dentra/esphome-tion/packages/persistent.yaml
  ## Uncomment next line to enable additional connect switch useful for debugging.
  connect: github://dentra/esphome-tion/packages/connect.yaml
  ## Uncomment next line to enable additional debug information.
  debug: github://dentra/esphome-tion/packages/debug.yaml
  ## Uncomment next line to if you wish to compile via ESP IDF.
  #idf: github://dentra/esphome-tion/packages/idf.yaml
  # Common configuration parameters. Do not comment or remove!
  common: github://dentra/esphome-tion/packages/common.yaml

# WiFi hotspot configuration.
# Comment next 3 lines to enable manual WiFi conviguration via access point.
wifi:
  ssid: $wifi_ssid
  password: $wifi_password

ble_client:
  mac_address: $tion_mac_address
  id: tion_ble_client

# Virtual port configuration
vport:
  - platform: tion_ble_3s
    id: tion3s_ble_vport
    ble_client_id: tion_ble_client
    # Optional, How often query device state. Default: 60s.
    update_interval: 60s
    # Optional, Do not disconnect after receiving state. Default: false.
    persistent_connection: false
    # Optional, Timeout to reconnect if no state answer received. Default: 15s.
    state_timeout: 15s
    # Optional, Sena a pair command after every connect (experimental feature)
    #experimental_always_pair: false
    # Optional, Allow to disable other BLE device scanning when breezer is already connected.
    disable_scan: false

# Main component configuration.
climate:
  - platform: tion_3s
    id: tion_climate
    vport_id: tion3s_ble_vport
    name: "$name"
    # Optional, Buzzer control switch.
    buzzer:
      name: "$name Buzzer"
    # Optional, Temperatire of air before heater, °C (celsius).
    outdoor_temperature:
      name: "$name Outdoor Temperature"
    # Optional, Current productivity, m³/h (cubic meters per hour).
    productivity:
      name: "$name Productivity"
    # Optional, Filter time left counter, days.
    filter_time_left:
      name: "$name Filter Time Left"
    # Optional, Air Intake control.
    air_intake:
      name: "$name Air Intake"
    # Optional, Tion firmware version.
    version:
      name: "$name Version"
    # Optional, Enable presets. Default: <empty>
    # Awailable configurable presets: home, away, boost, comfort, eco, sleep, activity.
    # Defaults:
    #   home    : { fan_speed: 2, target_temperature: 20, mode: heat }
    #   away    : { fan_speed: 1, target_temperature: 10, mode: fan_only }
    #   boost   : { fan_speed: 6, target_temperature: 10, mode: fan_only }
    #   comfort : { fan_speed: 2, target_temperature: 23, mode: heat }
    #   eco     : { fan_speed: 1, target_temperature: 16, mode: heat }
    #   sleep   : { fan_speed: 1, target_temperature: 18, mode: heat }
    #   activity: { fan_speed: 3, target_temperature: 18, mode: heat }
    # Boost is a special preset that runs for a period defined by boost_time and then switches back to the previous preset.
    presets:
      # enable and override "home" preset
      home: { fan_speed: 2, target_temperature: 20, mode: heat }
      # enable "away" preset
      away:
      # enable "boost" preset
      boost:
      # enable and override "sleep" preset
      sleep: { fan_speed: 1, target_temperature: 18, mode: heat }
    # Optional, configure boost time
    boost_time:
      name: $name Boost Time
    # Optional, display remaining boost time
    boost_time_left:
      name: $name Boost Time Left

button:
  - platform: template
    name: "$name Pair"
    on_press:
      lambda: id(tion3s_ble_vport).pair();
    entity_category: config
  ## Uncomment next 5 lines to enable switch for enable reset pair functionality.
  - platform: template
    name: "$name Reset Pair"
    on_press:
      lambda: id(tion3s_ble_vport).reset_pair();
    entity_category: config
```

fishenebelny, 2023-02-05:
Проблема повторяется. Перезагрузка по кнопку резет не помогает. Пока только повторное сопряжение помогло.

dentra, 2023-02-13:
Статус 133, говорит о невозможности соединения и, к сожалению, очень расплывчатый и может говорить о чем угодно. Вплоть до нехватки питания или слишком большого расстояния от esp до устройства. В вашем случае получается это было отсутствующее сопряжение. Проблема решена как понимаю?

fishenebelny, 2023-02-13:
Нет, проблема пока существует. Недели через 2 попробую перевести все это дело на esp-idf. Может бытт поможет.
А так сдела сопряжение повторно и опять заработало около 8 часов и опять сэвсе пропало.

fishenebelny, 2023-03-08:
Попробовал все через esp-idf. Результат такой же.
Попробовал постоянное подключение - тоже в какой то момент все зависает и перестает работать.
Может кто-то рассказать про свою стабильно работающую систему? 
У меня 2 Тиона 3S на прошивке 33. Модули  ESP32 с Алиэкспресс.

У кого то есть что то аналогичное, что работает стабильно? Можете поделиться конфигом?

dentra, 2023-06-22:
На этапе активного тестирования находится версия с «железным» подключением. В таком виде стабильность будет максимальная.

dentra, 2023-07-03:
версия с uart или "железным" подключением зарелижена. подключение очень простое и не требует навыков пайки.