dentra/esphome-tion #11: Просто наблюдение - в обновлённом приложении Тиона есть обновление прошивки

u17194907425, 2023-12-19:
Сабж.

Попытался поискать список изменений, но не нашёл. Написал им на почту в службу поддержки.
Интересно, кто-нибудь обновлялся и всё-ли после обновления работает.

AleksandrKozin, 2023-12-19:
Что-нибудь ответили?

dentra, 2023-12-19:
Я не имею никакого отношения к Tion. 
Весь проект построен на реверс инжиниринге протокола обмена собственных устройств и устройств сообщества.

didjeru, 2023-12-19:
Возможно это для прошивки Magic Air?

u17194907425, 2023-12-20:
> Я не имею никакого отношения к Tion.
> Весь проект построен на реверс инжиниринге

Это и не подразумевалось. Мысль была в том, чтобы проинформировать сообщество и поделиться, если есть чем. Вдруг, кто уже обновился и может подтвердить, что на работу с [esphome-tion](https://github.com/dentra/esphome-tion) это обновление не влияет.

> Что-нибудь ответили?
Пока ничего. Сегодня с утра (то есть через сутки) пришёл автоматический ответ, что моему обращению присвоен номер.

> Возможно это для прошивки Magic Air?
Возможно, но нет.

u17194907425, 2023-12-20:
В теории, никакой регрессии от обновления быть не должно. Я исхожу из того, что уважаемый @dentra зареверсил интерфейс пульта и Magic Air (если мы говорим про блюпуп). Если они его изменят, то пульты просто перестанут работать.

И в то же время, интересно, что же они там поменяли.

dentra, 2024-01-09:
В теории все должно продолжить работать иначе отвалятся родные пульты, которые работают по тому же самому протоколу.

dentra, 2024-08-13:
думаю уже не актуально

closed 2024-08-13 by dentra