dentra/esphome-tion #13: Compilation error on captive_portal

u17194907425, 2024-01-03:
Try compiling the recommended [profile for Tion 4S BLE](https://github.com/dentra/esphome-tion/blob/master/tion-4s.yaml) on ESPHome 2023.12.5

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/tion-bug.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tion-4s (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing etlcpp/Embedded Template Library @ 20.35.8
INFO Installing etlcpp/Embedded Template Library @ 20.35.8
Unpacking  [####################################]  100%
Library Manager: Embedded Template Library@20.35.8 has been installed!
INFO Embedded Template Library@20.35.8 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.0.1
INFO Installing esphome/AsyncTCP-esphome @ 2.0.1
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.0.1 has been installed!
INFO AsyncTCP-esphome@2.0.1 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- Embedded Template Library @ 20.35.8
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling .pioenvs/tion-4s/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/ble_client/automation.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/ble_client/ble_client.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/ble_rssi/ble_rssi_sensor.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/tion-4s/src/esphome/components/climate/climate.cpp.o
src/esphome/components/captive_portal/captive_portal.cpp: In member function 'void esphome::captive_portal::CaptivePortal::start()':
src/esphome/components/captive_portal/captive_portal.cpp:51:48: error: invalid cast from type 'esphome::network::IPAddress' to type 'uint32_t' {aka 'unsigned int'}
   this->dns_server_->start(53, "*", (uint32_t) ip);
                                                ^~
*** [.pioenvs/tion-4s/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
========================= [FAILED] Took 18.33 seconds =========================
```

u17194907425, 2024-01-06:
I would suspect the `ESPAsyncWebServer-esphome @ 2.1.0` version. If I do compile a clean ESP32 yaml with captive portal then the dependency graph shows `ESPAsyncWebServer-esphome @ 3.1.0`

dmdmvl, 2024-01-09:
> You need to replace package version to master. Only for package!

Changed reference to master
```
packages:
  tion:
    url: https://github.com/dentra/esphome-tion
    ref: master # cant use substitutions here
```
It compiles now

closed 2024-01-15 by dentra