dentra/esphome-tion #14: Не хватает сенсоров.

man55, 2024-01-17:
В интеграции отсутсвует сенсор скорости вентилятора. Из-за этого невозможно посмотреть историю скорости вентилятора.
Можно его добавить?

п.с. Так же на страничке интеграции устройства не отображается сенсор входящей температуры(outdoor temperature), хотя такой entity присутсвует.

[image]
[image]

dentra, 2024-01-17:
Сенсор скорости вентилятора можно сделать самостоятельно, пример чуть позже приведу.

Состояние работы обогревателя видно в истории климата. Или надо как-то по другому?

Air Intake - это и есть состояние заслонки

man55, 2024-01-18:
В истории вижу только две температуры: таргет и текущую. Поэтому сложно понять что происходило с бризером последние сутки, был ли включен обогреватель или нет, если да, то в какой период. Так же не видно на какой скорости он работал в какое время.
Очень сложно сделать автоматизацию и проверить её работу из-за отсутсвия таких элементарных сенсоров.

dentra, 2024-01-18:
добавьте в свой конфиг:

```yaml
climate:
  - platform: tion_3s
    id: !extend tion_climate
...
    on_state:
      - lambda: "id(tion_fan_speed).publish_state(x.custom_fan_mode.has_value() ? *x.custom_fan_mode.value().c_str() - '0' : 0);"
      - lambda: "id(tion_heating).publish_state(x.action == climate::CLIMATE_ACTION_HEATING);"

sensor:
  - platform: template
    id: tion_fan_speed
    name: Fan Speed
binary_sensor:
  - platform: template
    id: tion_heating
    name: Heating
```

dentra, 2024-01-18:
а вообще обогрев так виден:
[image]

man55, 2024-01-18:
Спасибо. С состоянием обогревателя понял.
Фан спид заработал, только отображает почему-то с десятичным знаком
[image]

dentra, 2024-01-18:
добавьте в сенсор accuracy_decimal: 0 или измените точность прямо в HA, так же можете изменить иконку:
```yaml
sensor:
  - platform: template
    id: tion_fan_speed
    name: Fan Speed
    accuracy_decimal: 0 
    icon: mdi:fan
binary_sensor:
  - platform: template
    id: tion_heating
    name: Heating
    icon: mdi:heat-wave
```

man55, 2024-01-18:
Спасибо, работает. Ругнулось на ошибку, правильно accuracy_decimal**s**

dentra, 2024-01-19:
скопипастил криво 👼

closed 2024-01-19 by dentra