dentra/esphome-tion #23: Подключение Тиона 3S по uart

dennismora, 2024-02-02:
Здравствуйте, выложите пожалуйста кто-нибудь свой реально работающий tion-3s-uart.yaml   
для подключения Tion 3s  UART  по mqtt в спрутхаб, чтобы было понимание где и что менять, чтоб пробелы и вводы не напутать. Я новичок, а хочется разобраться

dentra, 2024-02-02:
В телеграм вроде разобрались?

dennismora, 2024-02-03:
Надеюсь да)) Еще не установил все, я может потом свой yaml выложу - мне кажется будет полезно, я бы был рад конкретному примеру, для новичков полезно, хотя тут наверное новичков мало

dennismora, 2024-02-04:
[tion-esp.json](https://github.com/dentra/esphome-tion/files/14157525/tion-esp.json)

Итак это мой рабочий yaml для интеграции со спрутхабом на Тион S3, вцелом действительно несложно - прошивал плату через интеграцию ESP Home в Home assistant, там сам создался secret с данными моей вайфай сети, также убрал решетку в настройках mqtt и поменял их для брокера mqtt спрутхаб и наоборот поставил решетку для api, потому что для спрутхаба api не нужно. Еще из изменений в шаблоне поменял имя Tion-3s на Tion_3s без этого не прошивалось. Подключил микросхему по документации пдф с картинками и схемой подключения и закинул шаблон для spruthub отсюда https://github.com/l0rda/sprut-tion в папку на флешку для спрутхаба Sprut.hub\Templates\MQTT, не забудьте перезагрузить шаблоны. Еще раз большое спасибо создателям всех шаблонов

dennismora, 2024-02-04:
Не знаю насколько корректно размещать это здесь, но я записал подробное видео по настройке  всего этого, вдруг кому пригодится https://dzen.ru/video/watch/65bfecb7d46961389f809a2b

closed 2024-02-13 by dentra

xtinkster, 2026-09-19:
Вот полная инфа как я это делал - https://tinkster.dev/project/turning-a-tion-3s-breezer-into-a-co2-controlled-smart-supply-ventilator-with-esphome