dentra/esphome-tion #24: Суммарное дневное потребление растёт при выключенном бризере

dentra, 2024-02-02:
> а так и задумано что суммарное дневное потребление растёт при выключенном бризере? глянул по диагонали код, там вроде нет поверки на то, включен бризер или нет, то есть похоже продолжает считать по установленной скорости

dentra, 2024-02-02:
Необходимо добавить проверку на режим off в пакет tion_energy.yaml

dentra, 2024-02-20:
В мастере. Можно использовать

Dexter-Lucky, 2024-06-24:
Эта проблема на текущий момент имеет место быть как на 4s_ble, так и на lite.

Dexter-Lucky, 2024-08-09:
Залил код из ветки next.
Проблема сохранилась. При этом как-то не корректно отображается мощность вентилятора как на 4s, так и на Lite. Режим ожидания показывает корректно, как в описании, 0,73 Вт, а вот при работе, мощность отображается в 10 раз больше.
Ниже скриншоты в выключенном состоянии (0,73 Вт)
[image]
на третьей скорости (162 Вт)
[image]

На выходных  постараюсь измерить реальное потребление бризера Lite и 4s . Результаты пришлю сюда.

dentra, 2024-08-11:
10 кратное потребление поправил. буду благодарен за замеры

Dexter-Lucky, 2024-08-12:
Сделал на выходных замеры, результаты во вложении.
Сразу оговорюсь, что измерял прибором с точностью до  1 Вт.
По бризеру 4s результаты очень сильно похожи на уже вшитые данные, за исключением потребления на 6 скорости. Там потребление  получилось 35 Вт.
По Lite значения ожидаемо получились сильно ниже, чем у 4s. Сначала даже не поверил,дважды перепроверял  все режимы.
[Бризеры мощность.xlsx](https://github.com/user-attachments/files/16590267/default.xlsx)

dentra, 2024-08-13:
по 4S я константы усреднил

dentra, 2024-08-13:
В коммите [6493316](https://github.com/dentra/esphome-tion/commit/64933161337937e1b733b9d63f6fefc1b6225d17) добавил возможность переопределить встроенные значения. Ниже пример как это можно сделать, значение потребляемой мощности необходимо умножить на 100, например переопределяем 6 скорость на 35 Вт:
```yaml
esphome:
  platformio_options:
    build_flags:
      - -DTION_4S_MAX_FAN_POWER6=3500
```

dentra, 2024-08-13:
Решено

closed 2024-08-13 by dentra