dentra/esphome-tion #25: Нагрев даже в режиме fan_only

WeslyG, 2024-02-12:
После какого то из обновлений, заметил большое потребление тиона в энергомониторинге.
Полез разбираться, выяснил, что теперь даже в режиме "проветривание" если температура ниже нуля, то автоматом включается подогрев до нуля. Есть ли вариант отключить это поведение? Да я хочу холодный воздух в квартиру без обогрева, если он мне понадобиться, я переведу его в нагрев сам. 
Tion 4s ble версия плагина 2024.1.3

[image]
[image]

dentra, 2024-02-13:
если температура входящего воздуха ниже 0, то обогрев включается автоматически самим тионом и это поведение изменить невозможно. это встроенная защита от обмерзания.

WeslyG, 2024-02-13:
Воу, не знал этого, вроде раньше оно иначе работало, или просто статистику другую рисовало... Ну раз так, то ладно, спасибо за обьяснение

closed 2024-02-13 by WeslyG