dentra/esphome-tion #28: Периодические зависания

QAstron, 2024-03-13:
Установил прошивку на стик Lilygo T-Dongle S3, подключил к бризеру, всё было в порядке. Устройство проработало полутора суток, потом стик "отвалился". В сети недоступен, на команды из HA не реагирует, данные не выдаёт. При этом сам бризер продолжает работать и управляться через пульт и родное приложение. Восстановление работоспособности только через снятие питания (выключение из розетки или изъятие донгла из бризера и повторное его подключение). После сброса по питанию работоспособность восстановилась, бризер проработал ещё пять суток, и сегодня ситуация повторилась. Вынес стик на удлинителе наружу бризера чтобы промониторить возможный перегрев - такового не обнаружено. Остаётся две возможных причины: бракованный стик либо слабое/зашумлённое питание от самого бризера. Испытание в процессе, заказан ещё один стик, пока в пути. Можно ли на уровне кода добавить ватчдог (если его ещё нет конечно), который бы автоматически ресетил плату при её зависании? Хотелось бы также знать, сталкивался ли кто-либо ещё с такой же проблемой как у меня, и если да, то как её решали? Да, можно подключить бризер через управляемую розетку и дёргать питание каждый раз, когда плата перестаёт отвечать, но это имхо костыль.

dentra, 2024-03-14:
А вы случайно не меняли параметр heartbeat

QAstron, 2024-03-14:
> А вы случайно не меняли параметр heartbeat_interval?

Нет, не менял. В прошивку не вмешивался, там всё по умолчанию.

dentra, 2024-03-14:
Я пока первый раз слышу о такой проблеме, стики от Лилиго были очень хорошего качества. Возможно конечно производитель что-то «оптимизировал» в железе. Дело в том, что стик не проектировался под такой кейс использования и работает не совсем в штатном режиме, но за счет дополнительных резисторов работает как надо.

У вас новая версия стика, с дополнительным разъемом?

Если это то, что я предполагаю, то у меня была похожая ситуация летом, при перегреве стика, стик тоже переставал отдавать команды, но это было при 30 градусах на улице и всего пару раз.

Для начала, можно попытаться поймать ошибку по воздуху, если не будут уходить heartbeat-ы, скорее всего получим пропадание связи, бризер отключит порт.

QAstron, 2024-03-15:
Да, сбоку платы стика присутствует небольшой белый разъём, видимо, это как раз новая ревизия. В логах ничего "криминального" нет. Есть Хартбит, раз в 30 секунд бризер передаёт телеметрию, в общем на первый взгляд всё хорошо. Разобрал корпус, чтобы проконтролировать температуру непосредственно на чипах. Температура порядка 40-45 градусов (при температуре среды 25 градусов). К сожалению более точно назвать температуру не могу, т. к. куда-то подевал термопару мультиметра, а пирометр показывает с погрешностью из-за малого размера платы. В любом случае, перегрева нет, плата чуть тёплая, температура за пару часов не поменялась, значит дело не в нём. Буду наблюдать дальше. Когда придёт ещё один стик, появится возможность сравнить стабильность работы на обоих экземплярах.

dentra, 2024-03-18:
не не не, харбит - это отдельная команда, она в логе так прямо и видна, должна быть не реже чем раз в 5 секунд, раз в 30 секунд опрашивается состояние бризера

Blckb, 2024-08-23:
Кстати, возможно это другая проблема, но у меня с какой-то версии эти же стики, если верить показаниям температурного датчика, разогреваются до 115-120 градусов. Два бризера и одинаковая история. В комнате где кондиционер, греется сильно меньше, но и в другой комнате температура воздуха выше 25 градусов не поднимается. Аналогично я не менял никаких настроек. Стик установлен под крышкой корпуса Tion 4S, но вытащил резиновые заглушки отсеков где порты. Это немного уменьшило температуру. Версия стика с дисплеем в пластиковом корпусе. Возможно у него реально проблемы с отводом тепла, либо же в коде что-то что так нагружает расчётами.
Через 2-3 суток нагрева бризер выключается. Сейчас вот думаю без дисплея заказать стики и отключить функционал который не использую.

dentra, 2024-12-06:
В декабрьском релизе кардинально уменьшен нагрев стика. Посмотрите как будет себя вести

dentra, 2025-02-07:
еще актуально?

dentra, 2026-10-23:
Закрываю за отсутствием ответа. Нагрев стика снижен в релизе 2024.12. Если проблема повторится на актуальной версии — откройте новую задачу с логами.

closed 2026-10-23 by dentra