dentra/esphome-tion #30: Ошибка при заливке на ESP32

Dexter-Lucky, 2024-05-28:
Добрый день!
При заливке на плату выскакивает ошибка:
`> In file included from src/esphome/components/tion_lt/vport/tion_lt_vport.cpp:2:
> src/esphome/components/tion_lt/vport/tion_lt_vport.h:10:36: error: 'TionBleIO' in namespace 'esphome::tion' does not name a template type
>  using TionLtBleIO = esphome::tion::TionBleIO<dentra::tion::TionLtBleProtocol>;
>                                     ^~~~~~~~~
> src/esphome/components/tion_lt/vport/tion_lt_vport.h:12:52: error: expected template-name before '<' token
>  class TionLtBleVPort : public TionVPortBLEComponent<TionLtBleIO, TionLtBleIO::frame_spec_type> {
>                                                     ^
> src/esphome/components/tion_lt/vport/tion_lt_vport.h:12:52: error: expected '{' before '<' token
> src/esphome/components/tion_lt/vport/tion_lt_vport.h:12:52: error: expected unqualified-id before '<' token
> src/esphome/components/tion_lt/vport/tion_lt_vport.cpp:9:34: error: invalid use of incomplete type 'class esphome::tion::TionLtBleVPort'
>  void TionLtBleVPort::dump_config() { TION_VPORT_BLE_LOG("Tion LT BLE"); }
>                                   ^
> In file included from src/esphome/components/tion_lt/vport/tion_lt_vport.cpp:2:
> src/esphome/components/tion_lt/vport/tion_lt_vport.h:12:7: note: forward declaration of 'class esphome::tion::TionLtBleVPort'
>  class TionLtBleVPort : public TionVPortBLEComponent<TionLtBleIO, TionLtBleIO::frame_spec_type> {
>        ^~~~~~~~~~~~~~
> *** [.pioenvs/breez-bed/src/esphome/components/tion_lt/vport/tion_lt_vport.cpp.o] Error 1`

Пробовал как версию BLE для 4S, так и для Lite.
Ниже прикладываю конфигурацию:
[substitutions.txt](https://github.com/dentra/esphome-tion/files/15473008/substitutions.txt)

dentra, 2024-06-25:
как понимаю, по дальнейшим ошибкам уже с работой, проблема была решена?

Dexter-Lucky, 2024-06-25:
Не совсем.
Прошивку удалось залить только после отключения одного пакаджа:
 - packages/persistent.yaml

dentra, 2024-08-13:
Воспроизводится на 2024.8.0?

dentra, 2026-10-24:
С 2024.1.3 структура компонентов и пакетов полностью переработана. Конфигурация LT BLE + persistent (packages/tion/persistent.yaml) собирается в CI-матрице начиная с 2026.10.x. Закрываю, если на актуальной версии проблема повторится - откройте новый issue с полным логом сборки.

closed 2026-10-24 by dentra