dentra/esphome-tion #34: Проблема с установкой после обновления на ESPHome 2024.6.1

alancik80, 2024-06-20:
После обновления ESPHome при попытке установки выпадает вот такое:
```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/my4s.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1057, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1093, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 947, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 937, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 774, in validate_config
    config = do_packages_pass(config)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 179, in do_packages_pass
    recursive_package = do_packages_pass(package_config)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 180, in do_packages_pass
    config = merge_config(recursive_package, config)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 101, in merge_config
    return merge(full_old, full_new)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 58, in merge
    ids = {
          ^
  File "/esphome/esphome/config_helpers.py", line 61, in <dictcomp>
    if (v_id := v.get(CONF_ID)) and isinstance(v_id, str)
                ^^^^^
AttributeError: 'EStr' object has no attribute 'get'
```

Конфигурацию брал стандартную: https://github.com/dentra/esphome-tion/blob/master/tion-4s-uart.yaml
Железка: LILYGO T-Dongle-S3 ESP32-S3

alancik80, 2024-06-20:
Сам же отвечаю.
Вылетает из-за этих опций в base.yaml и esp32_s3.yaml

```
  platformio_options:
    board_build.flash_mode: dio
    build_flags:
      - -std=gnu++17
    build_unflags:
      - -std=gnu++11
```

Dexter-Lucky, 2024-06-24:
Аналогичная проблема

scherbet, 2024-06-24:
Шью 4s_ble, проблема аналогична.

dentra, 2024-06-25:
проблема известная, уже пофикшено в esphome 2024.6.2

closed 2024-06-25 by dentra