dentra/esphome-tion #35: Пропала температура наружного воздуха

Dexter-Lucky, 2024-06-24:
Почему то пропала температура наружного воздуха. Даже в атрибутах устройства ее нет.
Это и на 4s_ble и на lite воспроизводится.
Код прикрепляю.
Так же на 4s пропала информация о сроке замены фильтра, она почему то отображается теперь в оставшемся времени турбо режима.
ESP Home 2024.5.5

dentra, 2024-06-25:
воспроизводиться ли эта проблема на бете (ветка next)? home assistant или spurt?

Dexter-Lucky, 2024-06-25:
К сожалению не знаю, как использовать бета версию.
На текущий момент не обновляю ESP Home, т.к. не могу обновить платы на новой версии.
Home  Assistant

dentra, 2024-07-03:
Вы хотели что-то прикрепить, но забыли. Покажите хоть скриншот

Dexter-Lucky, 2024-07-04:
Прикладываю скриншоты
[image]
[image]
[image]

dentra, 2024-07-04:
Я очень надеюсь, что переименование вы делаете именно в HA, а не конфигурации прошивки т.к. из имени генерируется уникальный идентификатор, который попадает в HA, а с русскими буквами все плохо к сожалению.

покажите еще пожалуйста ваш конфиг.

Dexter-Lucky, 2024-07-08:
Добрый день.
Тему можно закрывать.  При заливке оригинального конфига все показатели появились, и снова вернулась проблема с 4s ble (но это в отдельной теме).
Спасибо.

closed 2024-08-11 by dentra