dentra/esphome-tion #36: fatal error: esphome/components/ota/ota_component.h: No such file or directory

hertznsk, 2024-06-30:
После одного из обновлений ESPHome стала возникать ошибка.

По сравнению с примером конфигурации <https://github.com/dentra/esphome-tion/blob/2024.1.3/tion-4s-uart.yaml> были добавлены директивы ota чтобы иметь возможность обновлять по сети. Ранее это не вызывало проблем.

Можно ли с этим что-то сделать?

ESPHome `2024.6.4`

Конфигурация:

```yaml
#
# https://github.com/dentra/esphome-tion
#

substitutions:
  # name of the node.
  node_name: tion-4s
  # name for main climate entity and prefix for all others.
  friendly_name: $node_name
  # add suffix for node name to have one firmware for many devices.
  node_add_mac_suffix: "false"

packages:
  tion:
    url: https://github.com/dentra/esphome-tion
    ref: 2024.1.3 # cant use substitutions here
    files:
      ## Enable additional WiFi signal sensor. Comment next line to disable.
      - packages/wifi_signal.yaml
      ## Enable additional Uptime sensor. Comment next line to disable.
      - packages/uptime.yaml
      ## Enable additional Restart button. Comment next line to disable.
      - packages/restart.yaml
      ## Base climate definition. Do not comment or move or remove.
      - packages/tion_4s_climate.yaml
      ## Enable maximum target temperature to 30°C. Comment next line to disable.
      - packages/tion_temperature30c.yaml
      ## Enable heat_cool mode to workaraund when using Home Assistant turn_on service.
      # - packages/tion_enable_heat_cool.yaml
      ## Uncomment next line to enable additional energy sensor.
      - packages/tion_energy.yaml
      ## Common configuration settings. Do not comment or move or remove.
      - packages/base.yaml
      # change to esp32_s2.yaml/esp32_c3.yaml to use with ESP32-S2/ESP32-C3 chip or change to esp32.yaml to use with ESP32 chip.
      - packages/esp32_s3.yaml
      ## Uncomment next line to enable additional internal timers support.
      # - packages/tion_4s_timers.yaml
      ## Uncomment next line to enable monitor internal ESP temperature.
      - packages/esp32_temperature.yaml
      ## Uncomment next line to enable additional debug information.
      # - packages/debug.yaml

# Uncomment to enable MQTT connecion.
# WARNINIG: If you leave next api section for logging purpose
#           and do not use Home Assistant, you need to uncomment
#           reboot_timeout parameter.
# mqtt:
#   broker: !secret mqtt_borker
#   port: !secret mqtt_port
#   username: !secret mqtt_username
#   password: !secret mqtt_password

# Enable Home Assistant API connection. Please do not remove to keep logs functioning.
api:
  # Uncomment next line if you do not use Home Assistant.
  #reboot_timeout: 0s
  encryption:
    key: !secret api_key

[... 225 lines cut ...]
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/log.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-3s.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-4s.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-ble-3s.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-ble-lt.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-lt.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-uart-3s.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api-uart-4s.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/tion-api.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion-api/utils.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion_climate.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion_climate_component.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion_component.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion_fan.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion/tion_fan_component.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion_4s/climate/tion_4s_climate.o
Compiling .pioenvs/tion-4s/src/esphome/components/tion_4s_uart/tion_4s_uart.o
src/esphome/components/tion_4s_uart/tion_4s_uart.cpp:4:10: fatal error: esphome/components/ota/ota_component.h: No such file or directory
 #include "esphome/components/ota/ota_component.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/tion-4s/src/esphome/components/tion_4s_uart/tion_4s_uart.o] Error 1
========================= [FAILED] Took 207.85 seconds =========================
```

Blckb, 2024-07-01:
К сожалению такая же ошибка. 
ESPHome 2024.6.4
Обновляюсь с 2023.7.0 до 2024.1.3.
yaml поправил.

dentra, 2024-07-02:
Для нормальной сборки на esphome больше 2024.6.0, до выхода релиза, временно используйте ветку next

dentra, 2024-08-13:
Релиз 2024.8.0 поддерживает esphome 2024.6 и выше

closed 2024-08-13 by dentra