dentra/esphome-tion #39: 2024.8.0 режим persistent не отображается уровень сигнала ble

tarikptz, 2024-08-27:
При включении режима persistent не отображается уровень сигнала ble. В обычном режиме работает.

dentra, 2024-10-04:
попробуйте добавить в конфиг, в секцию `sensor`, например, в начало, чтобы получилось примерно так:
```yaml
sensor:
 - platform: ble_client
   type: rssi
   ble_client_id: tion_ble_client
   name: BLE RSSI
   entity_category: diagnostic
```

dentra, 2026-10-24:
Исправлено в ветке `next`: пакет `rssi_signal` теперь показывает уровень сигнала и при постоянном подключении. Без постоянного подключения RSSI, как и раньше, берётся из рекламных пакетов бризера, а при постоянном подключении - из самого соединения, раз в 20 с.

Если вы добавляли сенсор `ble_client` с `type: rssi` из моего предыдущего комментария, удалите его - он больше не нужен.

ВАЖНО: Пакет рассчитан на конфиг из репозитория: vport должен иметь `id: tion_ble_vport`.

Отпишитесь, пожалуйста, заработало ли.

dentra, 2026-10-24:
Если будет работать не так как ожидается, заведите новую задачу

closed 2026-10-24 by dentra