dentra/esphome-tion #45: Группировка устройств climate и fan

Yiivgeny, 2024-11-20:
Возможно ли сгруппировать climate и fan в одно устройство? 

Так делает, например, стандартная midea_ac из ESPHome. Это позволяет более удобно управлять устройством при экспорте в HomeKit.

Скриншот как это выглядит для Midea
[image]

dentra, 2024-11-24:
Используйте только климат, эффект будет такой же

dentra, 2024-11-24:
Фан, отдельная сущность, кому климат не нравится

Yiivgeny, 2024-11-25:
1. Удалил устройство вентилятора из HomeKit
2. Пересобрал прошивку без fan (перезапустил HA)
3. Удалил климат из HomeKit (перезапустил HA, убедился что устройство исчезло из HomeKit)
4. Добавил климат в HomeKit (перезапустил HA)

 Ни после какого  пункта контрол не стал таким как на скриншоте

Yiivgeny, 2025-02-06:
Попробую разобраться в причине, спасибо

closed 2025-05-29 by dentra

Yiivgeny, 2025-05-29:
Поправилось или просто закрыли?