dentra/esphome-tion #46: Проблема с включением persistent connection

Yiivgeny, 2024-11-20:
Использую самую свежую на текущий момент next прошивку из [dc1d147](https://github.com/dentra/esphome-tion/commit/dc1d14779d41d7899d11d1d14ce7ae92985f89b1)

- Изолированно `persistent_connection=true` в vport работает корректно
- Добавление переключателя персистентного соединения (`packages/persistent.yaml`) тоже работает исправно

Но при использовании обоих этих пунктов, значение по умолчанию vport не работает. То есть после перезагрузки дефолтное значение из конфига не используется (точнее видимо сбрасывается). Иными словами невозможно получить после перезагрузки персистентное соединение если хочется иметь еще и переключатель

dentra, 2025-02-06:
исправлено, до релиза доступно в ветке next

Yiivgeny, 2025-02-06:
Проверил. Спасибо большое!

closed 2026-10-23 by dentra