dentra/esphome-tion #48: [3S] Log spam regression between 2024.8.0 and 2024.12.0

dshcherb, 2024-12-27:
There seems to be a regression between `2024.8.0` and `2024.12.0`: the module running esphome starts spamming many messages per second like below.

- HW setup: Tion 3s <- m5stack C3U <- original BLE module
- The MagicAir station isn't connected (unfortunately it enters a boot loop and doesn't get reset properly so I cannot confirm whether this has any effect - I just powered it off for now).

The only config change I did is this and reverting back results in normal logging:
```
diff --git a/tion-3s-c3u.yaml b/tion-3s-c3u.yaml
index 707fafe..241ce12 100644
--- a/tion-3s-c3u.yaml
+++ b/tion-3s-c3u.yaml
@@ -24,7 +24,7 @@ substitutions:
 packages:
   tion:
     url: https://github.com/dentra/esphome-tion
-    ref: 2024.8.0 # cant use substitutions here
+    ref: 2024.12.0 # cant use substitutions here
     files:
       ## You can enable or disable some of the features listed below.
       ## The disabled feature starts with # symbol.
```

The messages occur very rapidly as soon as I flash `2024.12.0` and on `2024.8.0` they appear every minute.
```
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[09:32:24][D][tion-api-3s:206]: Request Timers
[... 173 lines cut ...]
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:115]: Response Timers: 11.00.08.00.08.00.08.00.00.00.00.00.00.00.00.00.00 (17)
[09:32:25][D][tion-api-3s:206]: Request Timers
[09:32:25][D][tion-api-3s:206]: Request Timers
```

dentra, 2025-01-09:
Извиняюсь, но немного отойду от темы задачи.

Подскажите пожалуйста по подключению вашей esp. Это первая esp32 которую я вижу совместно с 3S. Питаете ее от пина 5В или 3.3В? Колодка на ней распаяна?

dshcherb, 2025-01-15:
> Поправлено. Пока можно использовать ветку next

Большое спасибо! Попробовал `next` - всё ок.

Заодно спасибо за [фикс ](https://github.com/dentra/esphome-tion/commit/ed2673e01f0ef18355c1b70929e618087bd5b2e7) авто режима - раньше и правда сбрасывался на ручной.

Попробовал с [внешним датчиком](https://devices.esphome.io/devices/M5Stack-AirQ) с простыми правилами (ПИД-регулятор протестирую тоже):

```
# ...
sensor:
  - platform: homeassistant
    id: airq_co2
    name: "CO2 AirQ Sensor"
    entity_id: sensor.airq_co2
    device_class: carbon_dioxide
    state_class: measurement
# ...
tion:
  type: 3s
  id: tion_api
# ...
  auto:
    co2: airq_co2
    lambda: |-
      if (x > 800) return 4;
      if (x > 700) return 3;
      if (x > 600) return 2;
      return 1;
# ...
climate:
  - platform: tion
    id: tion_climate
    name: "Tion 3S"
    enable_heat_cool: True
    enable_fan_auto: True
```

1) Дыхание на датчик CO2 -> ~2200 PPM - увеличение скорости.
```
[03:17:53][D][homeassistant.sensor:024]: 'sensor.airq_co2': Got state 2197.00
[03:17:53][D][sensor:093]: 'CO2 AirQ Sensor': Sending state 2197.00000  with 1 decimals of accuracy
[03:17:53][D][tion_api_component:088]: Starting batch update: 200 ms
[03:17:53][D][tion-api-3s:151]: Request State Get
[03:17:53][D][tion_api_component:024]: Write out batch changes
[03:17:53][D][tion-api:205]: New fan speed 1 -> 4
```

2) Снижение скорости по мере уменьшения показаний:
```
[03:20:53][D][homeassistant.sensor:024]: 'sensor.airq_co2': Got state 621.00
[03:20:53][D][sensor:093]: 'CO2 AirQ Sensor': Sending state 621.00000  with 1 decimals of accuracy
[03:20:53][D][tion_api_component:088]: Starting batch update: 200 ms
[03:20:53][D][tion-api-3s:151]: Request State Get
[03:20:53][D][tion_api_component:024]: Write out batch changes
[03:20:53][D][tion-api:205]: New fan speed 4 -> 2
```
```
[03:21:53][D][homeassistant.sensor:024]: 'sensor.airq_co2': Got state 575.00
[03:21:53][D][sensor:093]: 'CO2 AirQ Sensor': Sending state 575.00000  with 1 decimals of accuracy
[03:21:53][D][tion_api_component:088]: Starting batch update: 200 ms
[03:21:53][D][tion-api-3s:151]: Request State Get
[03:21:53][D][tion_api_component:024]: Write out batch changes
[03:21:53][D][tion-api:205]: New fan speed 2 -> 1
```

Отличная замена сломавшемуся Magic Air.

dentra, 2025-01-21:
Т.е. логика получается 5В там, доже с учетом того что питание может быть 3.3В?

У m5stack хорошая схемотехника и если есть отдельная нога RX, то она должна быть защищена резистором, в целом этого должно быть достаточно, чтобы не спалить проц и все работало без сбоев.

dentra, 2025-01-21:
А usb dis нужен в определенных случаях чтобы перевести usb в режим uart, это не связано с напряжением питания. У лилиго есть защитные резисторы на входе usb поэтому он «выживает»

dshcherb, 2025-01-22:
> Т.е. логика получается 5В там, даже с учетом того что питание может быть 3.3В?

У C3U логика 3.3В, но можно подать 5В на 5V пины и они регулятором будут снижены до 3.3В. Соответственно RX и TX тоже на 3.3 уровнях только.

Вот так соединил в итоге, думаю на схеме нагляднее будет:

[image]

> если есть отдельная нога RX, то она должна быть защищена резистором, в целом этого должно быть достаточно, чтобы не спалить проц и все работало без сбоев

По схеме не понял есть ли какая-то защита или нет и в ESP32-C3 datasheet-е тоже явного указания не нашел на 5V-tolerance. 
https://static-cdn.m5stack.com/resource/docs/products/core/stamp_c3u/stamp_c3u_sch_01.webp
https://docs.m5stack.com/en/core/stamp_c3u

Решил просто перестраховаться, т.к. конвертер уровней относительно дешевый.

> А usb dis нужен в определенных случаях чтобы перевести usb в режим uart, это не связано с напряжением питания. У лилиго есть защитные резисторы на входе usb поэтому он «выживает»

Да, понимаю, я посмотрел, что в лилиго есть 5V-tolerance для пинов, используемых под USB и они тем самым подходят для использования под UART. Хорошая идея для плат, где есть 5V-tolerance у USB пинов.

На D- D+ [уровень 3.3В](https://en.wikipedia.org/wiki/USB_communications#Electrical_specification) используется - просто решил не рисковать этими пинами, если там нет защиты.

dentra, 2025-01-26:
Все верно, 5V tolerance есть только у 8266 серии, все Esp32 работают только с 3.3В.

dentra, 2025-01-26:
Спасибо за схему, с вашего позволения размещу ее в репозитории по железу?

dshcherb, 2025-01-28:
> Спасибо за схему, с вашего позволения размещу ее в репозитории по железу?

Да, без проблем.

> Только на вашей схеме перепутаны названия. Прокси это BLE модуль.

Пардон, перепутал.

[image]

> А эта модель c3u уже с колодкой или паять надо?

Эту паял, но можно взять M5Stamp S3 с уже припаянными выводами: https://shop.m5stack.com/products/m5stamps3-with-2-54-header-pin

Брал C3U, т.к. думал, что у C3 будет энергопотребление поменьше, чем в S3, но надо бы измерить как оно фактически.

closed 2025-01-28 by dshcherb