dentra/esphome-tion #54: После пропадания wifi теряется связь с бризером

bazuchan, 2025-04-09:
В логах следующее:
`[14:34:58][D][tion-api-4s:294]: Request Heartbeat
[14:34:59][D][sensor:094]: 'WiFi Signal': Sending state -90.00000 dBm with 0 decimals of accuracy
[14:35:03][D][tion-api-4s:227]: Request Device info
[14:35:03][D][tion-api-4s:232]: Request State
[14:35:03][D][tion-api-4s:294]: Request Heartbeat
[14:35:06][W][tion_api_component:074]: State was not received in 3.0 s
[14:35:08][D][tion-api-4s:294]: Request Heartbeat
[14:35:13][D][tion-api-4s:294]: Request Heartbeat
[14:35:18][D][tion-api-4s:227]: Request Device info
[14:35:18][D][tion-api-4s:232]: Request State
[14:35:18][D][tion-api-4s:294]: Request Heartbeat
[14:35:21][W][tion_api_component:074]: State was not received in 3.0 s`

dentra, 2025-05-29:
а стик так и продолжает работать, только от бризера не приходит состояние, бризер не отлючает ему питание?

bazuchan, 2025-05-29:
> а стик так и продолжает работать, только от бризера не приходит состояние, бризер не отлючает ему питание?

да

dentra, 2026-10-24:
актуальна ли еще проблема?

bazuchan, 2026-10-24:
На ветке next не актуальна

closed 2026-10-24 by bazuchan